Blueforcer/awtrix3 #4: Allow to jump to a specific page via MQTT

TheZoker, 2023-03-22:
First: Thanks for including my request that fast in the latest release! 🎉 
(Just one question: Would it me possible to support the auto transition also from MQTT? Thanks!)

Since I disabled the auto transition it would be nice to allow to switch to a certain page (build in or custom) via MQTT.

For example:
I want to show the clock by default. When My song in spotify changes, I want to switch to my custom page for x seconds, then back to the clock and then stay at the clock page.

Blueforcer, 2023-03-23:
I can implement it.
But for this purpose I think a notification would be the better choice to display your the song.

TheZoker, 2023-03-23:
Good point! Just tried it works very nice!
But I still think that this option could be useful. What do you think?

commit 7b5a4d3a (2023-03-23): v0.35

- moves transistion time and app time setting into onscreen-menu
- Text starts to scroll on the left side instead from right
- allows to switch to a page by name via mqtt
- custompages doesnt need to have integers as an id anymore, you can also use strings now.

closes #7
closes #4