Blueforcer/awtrix3 #9: Creating custom page not working

b-a-9-0, 2023-03-23:
- MQTT prefix changed to "awtrix" in web ui.
- Sending MQTT message to "awtrix/custom/test"

- Message:
{
    "text": "512 W",
    "icon": "52090",
    "repeat": 1,
    "rainbow": false,
    "duration": 10,
    "color": "#FFFFFF",
    "hold": false
}

- Icon is uploaded to folder ICONS.

-> Custom page is not generated nor shown on display.

-> the same message to topic "awtrix/notify" works!

Blueforcer, 2023-03-23:
Unfortunately, I can not reproduce. Works here with the same data flawlessly.
Any other informations i should know?

b-a-9-0, 2023-03-23:
MQTT message is generated with node red:
[image]

MQTT explorer:
[image]

b-a-9-0, 2023-03-23:
I currently only have the standard pages: time, date, temp, humidity, battery. Running version 0.33

Blueforcer, 2023-03-23:
please update to latest 0.36 first. I cannot find bugs in older firmwares anymore :)

b-a-9-0, 2023-03-24:
Works in 0.36!

closed 2023-03-24 by b-a-9-0

aptonline, 2023-03-31:
I'm seeing the same issue when testing the latest firmware (0.45). Not displaying custom pages with the following command:

Topic:
`awtrix_d670d0/custom/demo
`
Payload:
```
{
  "text": "Hello, AWTRIX Light!",
  "icon": "1",
  "rainbow": true,
  "duration": 10
}
```

Testing the same payload with `awtrix_d670d0/notify` **intermittently** works.  Testing by sending from Home Assistant and MQTTX where I can see the payload correctly delivered to MQTT,