Blueforcer/awtrix3 #16: Expose button action on HA

cacordina, 2023-03-26:
Hi, do you think it will be possible to expose the action done on button to home assistant.( ex left button : none , one click , double click , long click ) and maybe have the ability to desactivate the navigation feature for one button ( left or right ) except for configuration menu ? Personally, I don't really use button to navigate between screen and just have one ( left or right ) is enough so if I can use the other button to trigger action on home assistant , it can be a really nice feature.

cacordina, 2023-03-26:
I did a pull request with an implementation of this. It works but probably not the best way to implement it :).

Blueforcer, 2023-03-26:
How do you deactivate the navigation? In case you only want to use the buttons on HA, you still control the clock.
Also what should happen when you want to dismiss a notification? this will also trigger your HA and may do unwanted things at this time.

cacordina, 2023-03-26:
Right now , no action are mapped for the double click and long click for left and right button so I use this action. For the validate button, I agree this seems difficult to use it with HA.

Blueforcer, 2023-03-29:
i think we should discuss this in more depth. there is still a lot to it, why you can't just add this as a generic function. And we should decide if this function is really necessary to alienate awtrix purpose.
https://github.com/Blueforcer/awtrix-light/discussions

closed 2023-03-29 by Blueforcer