Blueforcer/awtrix3 #17: Feature: when LDR reading is low, only show clock

poblabs, 2023-03-26:
Similar to the other matrix clock out there, when in a dark room it would be nice to have the awtrix-light only show the clock at brightness 1. 

I've had some success with determining that if [BRIGHTNESS](
https://github.com/Blueforcer/awtrix-light/blob/main/src/PeripheryManager.cpp#L199) < 20 then the room is dark. Defining that in this area seems to have worked, but I'm stuck trying to figure out how to modify the ui.setApps(Apps) to only include clock, then revert to the list of apps when brightness is above 20.

Blueforcer, 2023-03-26:
But this shouldnt be as other clocks. 
Apart from that I have already developed awtrix 2 which really offers a ton of features.

AWTRIX LIGHT lives on your smarthome. And you already have the possibility to implement exactly that according to your ideas with only 2 MQTT messages.
It always sounds simple, but in the end it is truly a monster of an adjustment. One wants to display 10% brightness and the temperature, the other wants it according to time and not according to brightness. All had in the last 4 years awtrix development :)
Use the power of mqtt commands and create your wish exactly as you want it.

poblabs, 2023-03-26:
> 

That sounds good. Is it possible to have the LDR measurements available in MQTT? This way node red can help automate this

Blueforcer, 2023-03-26:
Oh, sorry I've forgot that awtrix light currently expose it's stats only to Homeassistant. I will implement a generic mqqt stat message in the next version!

poblabs, 2023-03-26:
Actually that would be perfect except it looks like the light sensor isn't working right? It's not in a dark room but shows 0 right now 

[image]

poblabs, 2023-03-26:
Nevermind I guess it is working. the lux value goes up when I put a lot of light on it. I'll need to experiment a bit more with this. 

Using home assistant is it possible to "lock" the awtrix light on a certain app?

Blueforcer, 2023-03-26:
next version will also expose the raw values in a json via mqtt.
Currently the converting to lux ist inaccurat since i dont know the used LDR sensor nor the resistors.. so its just a guess right now.
maybe with the raw values we can made a better calculation in future updates.

you can switch to a an app and then send another message wich turns the autotransition off :
https://blueforcer.github.io/awtrix-light/#/mqtt

Blueforcer, 2023-03-26:
I think the latest update will help you to build the automation as you like 
https://github.com/Blueforcer/awtrix-light/commit/689cba1c854f93099f9460953ae9badda7811b02

closed 2023-03-26 by Blueforcer