Blueforcer/awtrix3 #19: Add Device configuration URL for HA

tioan, 2023-03-26:
Please add Support for [Device configuration URL](https://developers.home-assistant.io/blog/2021/10/26/config-entity/#device-configuration-url) 
to Change the MQTT Info button into an Link to Webinterface of the device

current:

[image]

desired:

[image]

Blueforcer, 2023-03-26:
This is currently not possible with the used library.
https://dawidchyrzynski.github.io/arduino-home-assistant/documents/library/device-configuration.html#device-configuration

 i will try to exend  this lib by myself in a future update.

Blueforcer, 2023-04-09:
I will close this issue, until the lib gets an update. Feel free to open it again if its possible

closed 2023-04-09 by Blueforcer