Blueforcer/awtrix3 #20: Extend Entity categories with diagnostic

tioan, 2023-03-26:
Extend existing [Entity categories](https://developers.home-assistant.io/blog/2021/10/26/config-entity?_highlight=diagnostic#entity-categories)with diagnostic and add the following buttons into the category
* Battery (form Sensors)
* Restart
* Update or "Check for Update and apply"

Probebly some internal Information like
- Uptime
- Free Mem
- WiFi Signal

Blueforcer, 2023-03-26:
diagnostics are not part of the used libary

poblabs, 2023-04-09:
+1 for Restart button

Blueforcer, 2023-04-09:
Personally i don't see any sense to restart the clock from HA, there is no need for that.
Every additional HA entity consumes a lot of RAM because awtrix needs to hold several jsons for it. So it must be chosen wisely.

Blueforcer, 2023-04-09:
I will close this issue, until the lib gets an update. Feel free to open it again if its possible

closed 2023-04-09 by Blueforcer