Blueforcer/awtrix3 #24: Feature: Have a unique MQTT topic by default

TheGroundZero, 2023-03-28:
This is a minor _quality of life_ request, especially for people with less experience in this kind of setup.

The default MQTT prefix is `AwtrixLight`.
If you have multiple clocks, you need to manually change this prefix to ensure they are each discovered as individual devices.

I'd propose appending the 3 final (NIC specific) octets of the device's MAC address to this prefix to make it unique by default.
Similar as to how other IoT devices setup their hostname and/or SSID*.

So if the device's MAC address is `12:34:56:78:90:AB`, the MQTT prefix could `AwtrixLight-7890ab`.

(*) using this for the WiFi SSID would be another useful little change.

Blueforcer, 2023-03-28:
I need to think about it, because now the prefix is used for the DNS hostname, mqtt and homeassistant.
For the hostname i need to use something else, because noone can remember AwtrixLight-7890ab :)

TheGroundZero, 2023-03-28:
Having the same hostname for multiple clocks would result in issues with mDNS routing anyway.
My Shelly's all have unique hostnames based on their MAC. WLED also uses unique hostnames as you can have multiple WLED installations in 1 network.

Static hostnames could still be set via the router or people could bookmark the (reserved) IP for the device.

Blueforcer, 2023-03-28:
youre right.
I will implement it

commit c51c769c (2023-03-28): v0.42

- App transistion is now inacitve if there is only 1 app
- Adds bargraph to notify and customapps
- Every awtrix now gets a unique id for AP, MQTT and HA
- Adds firmware as HA sensor
- Adds wifi strength as HA sensor
- Adds ram usage as HA sensor
- Adds version as HA sensor
- Adds uptime as ISO 8601 as HA sensot
- HA discorvery now gets correct device classes
- fixes bug where date formats are not saved

closes #24
closes #23
closes #21
closed 2023-03-28 by Blueforcer