Blueforcer/awtrix3 #30: [Feature Request] Possibility to check update by mqtt

uiltonlopes, 2023-04-01:
With this feature we can set a schedule to check and update the device automatically

TheZoker, 2023-04-01:
Would also be nice, to expose the update as a HA update sensor with a button to update the firmware just like the shelly (or esphome) integration:
[image]

But maybe this is limited by the external library

Blueforcer, 2023-04-01:
> Would also be nice, to expose the update as a HA update sensor with a button to update the firmware just like the shelly (or esphome) integration: [image]
> 
> But maybe this is limited by the external library

Unfortunally not possible

Blueforcer, 2023-04-01:
https://github.com/Blueforcer/awtrix-light/commit/bc72069ae3c9032cb4b1c99ff83cfb761b3ccb58

closed 2023-04-01 by Blueforcer