Blueforcer/awtrix3 #38: [QUESTION] Memory Leakage issue?

smarthomejunkie, 2023-04-03:
# Question
If I look at the free ram history in Home Assistant, I see that the RAM is slowly diminishing.
[image]

I created several apps through MQTT in Home Assistant that I show on Ulanzi Pixel Clock. Every time when a value changes within the app, I publish a new MQTT message (MQTT.publish) to the clock with the same app name. Is this the correct approach, or should I update the app using the MQTT.update command?

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.47

smarthomejunkie, 2023-04-03:
I monitored this for a while now and it seems that every time when I send an MQTT publish message, the memory gets a bit lower. I'd expect that the memory stays at the same level when you send an MQTT publish message (to the same app).

Blueforcer, 2023-04-03:
Did you noticed something similar before 0.47?

smarthomejunkie, 2023-04-03:
I didn't really look into it before, but according to the stats it seems to be a new issue since 0.47

[image]

Blueforcer, 2023-04-03:
I will have a look at it before next update

Blueforcer, 2023-04-03:
What happened at the last peak in your history? Restart?

smarthomejunkie, 2023-04-03:
Yes, it spontanously restarted when I was sending multiple mqtt publish commands.

Blueforcer, 2023-04-03:
i think i found something. I made a benchmark the last hour at it seems pretty stable at >3000 messages
[image]

smarthomejunkie, 2023-04-03:
That looks promising!

Diginix, 2023-04-07:
My TC001 freezed today after 2 days running with 0.47. It shows the time with not changing seconds when it happend. It was not reachable over network and no hardware button worked anymore. Only left+right button hold for turning off and on helped.
Where I can monitor the RAM? Didn't found it in stats or webUI.

Blueforcer, 2023-04-07:
You can find it in Homeassistant sensor. 
But it's already 100% fixed for the next version

Diginix, 2023-04-07:
I use TC001 with iobroker as mqtt broker. I guess I can't retrieve RAM usage there, but didn't need it.

Blueforcer, 2023-04-07:
That's true, next version will also send it to /stats. But then ram issue is solved 😄

Diginix, 2023-04-08:
Today I got again a freezed TC001. I switch every 30 sec from time app to a custom app for 10 sec with outside temperature and dynamic icon related to the current weather. With this frequency it happens at least once per day for me. Would be great to have the bug fix soon. I can also test beta versions if you tell me how to install.

Blueforcer, 2023-04-08:
You can get the Testversions on my discord server.

Diginix, 2023-04-08:
In awtrix-light section I see only the forum
[image]
Could you guide me please where I can download it?

Blueforcer, 2023-04-08:
Have a look at the dev talk channel. I posted a new version today. You can update it via Webinterface

Diginix, 2023-04-08:
0.48 found
[image]

PR #48 (2023-04-09): 0.48
- Fixes some rendering issues in the GIF decoder.
- 32x8 gifs are possible
- New json key: "textCase"
- Slight color correction
- Http /api/loop shows the current apps in loop.
- Sends uptime to HA in seconds closes #45 
- /apps api improved, deleting and reordering apps should work a bit better. closes #41 
- Memory leak fixed, closes #38
commit 222d979e (2023-04-10): Merge pull request #48 from Blueforcer/development

- Fixes some rendering issues in the GIF decoder.
- 32x8 gifs are possible
- New json key: "textCase"
- Slight color correction
- Http /api/loop shows the current apps in loop.
- Sends uptime to HA in seconds closes #45 
- /apps api improved, deleting and reordering apps should work a bit better. closes #41 
- Memory leak fixed, closes #38
closed 2023-04-09 by Blueforcer