Blueforcer/awtrix3 #45: [BUG] Fix uptime sensor in Home Assistant

TheGroundZero, 2023-04-07:
# Bug report

## Describe the bug

From the Home Assistant Logs:
> Sensor sensor.awtrix_xxxxxx_uptime has device class `duration`, state class `None` and unit `None` thus indicating it has a numeric value; however, it has the non-numeric value: P0DT7H56M8S (<class 'str'>);
> Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.45

## To Reproduce
Steps to reproduce the behavior:

## Expected behavior
A clear and concise description of what you expected to happen.

Have a properly setup uptime sensor.

Send the boot time as a datetime object / iso datetime string and have HA calculate the uptime for you?

## Screenshots
If applicable, add screenshots to help explain your problem.

[image]

## Logs

```text
Sensor sensor.awtrix_dd2824_uptime has device class duration, state class None and unit None thus indicating it has a numeric value; however, it has the non-numeric value: P0DT7H56M8S (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
```

## Additional context
Add any other context about the problem here.

Blueforcer, 2023-04-07:
The time format  is an ISO 8601 duration format.
Tell me how else to send it

TheGroundZero, 2023-04-07:
Perhaps unit_of_measurement or state_class need to be configured correctly HA to recognise this format vs HH:MM:SS ?

PR #48 (2023-04-09): 0.48
- Fixes some rendering issues in the GIF decoder.
- 32x8 gifs are possible
- New json key: "textCase"
- Slight color correction
- Http /api/loop shows the current apps in loop.
- Sends uptime to HA in seconds closes #45 
- /apps api improved, deleting and reordering apps should work a bit better. closes #41 
- Memory leak fixed, closes #38
commit 222d979e (2023-04-10): Merge pull request #48 from Blueforcer/development

- Fixes some rendering issues in the GIF decoder.
- 32x8 gifs are possible
- New json key: "textCase"
- Slight color correction
- Http /api/loop shows the current apps in loop.
- Sends uptime to HA in seconds closes #45 
- /apps api improved, deleting and reordering apps should work a bit better. closes #41 
- Memory leak fixed, closes #38
closed 2023-04-09 by Blueforcer