Blueforcer/awtrix3 #46: [QUESTION] Lost custom app after Restart ESP

120318, 2023-04-07:
# Question
[image]
If I click the Restart ESP button, my custom app will lost. is it normal?

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.47

Blueforcer, 2023-04-07:
Custom "apps" are only temporary hold in RAM. Just send it again

closed 2023-04-07 by Blueforcer

120318, 2023-04-07:
> Custom "apps" are only temporary hold in RAM. Just send it again

Is it possible and are there plans to make them permanently stored？

schneekluth, 2023-04-15:
> > Custom "apps" are only temporary hold in RAM. Just send it again
> 
> Is it possible and are there plans to make them permanently stored？

Just use MQTT's retained messages feature for your custom apps. Your broker will store your custom apps and resends them when you restart your pixel clock. See https://github.com/Blueforcer/awtrix-light/issues/11

Example:
```yaml
- service: mqtt.publish
  data:
    qos: 2
    retain: true
    payload: >-
      {
        "text": "{{ states.sensor.exchange_rate_1_btc.state | round(0) }}",
        "icon": "bitcoin"
      }
    topic: pixelclock/custom/bitcoin
```

Blueforcer, 2023-04-15:
Very nice idea