Blueforcer/awtrix3 #53: [FEATURE REQUEST] Change font

TheFes, 2023-04-11:
# Feature Request

### Is your feature request related to a problem?

Nope

### Describe the solution / feature you'd like

I would be nice if we would have an option to upload fonts in a FONTS folder, which can then be used to display the text on the clock (maybe even on notification/app basis). The FONTS folder would contain ttf files with the respective fonts.

### Describe alternatives you've considered

Stick to the custom font

### Additional context

I know it's possible with ESPHoMatrix to use different fonts, but AWTrix is faster and has more options, so I'm not switching back just because of this :)

Blueforcer, 2023-04-11:
do you have any fonts, so i can play with?

TheFes, 2023-04-11:
you can use [Calcium.ttf](https://blakadder.com/assets/files/Calcium.ttf) or have a look at the bitfontmaker2 [gallery](https://www.pentacom.jp/pentacom/bitfontmaker2/gallery/) (although a lot of the fonts in that gallery are higher than 8 pixels)

Blueforcer, 2023-04-12:
Unfortunately it doesnt seem to be possible to load ttf fonts at runtime. We had to convert it and compile it with the firmware.

TheFes, 2023-04-12:
Okay, thanks for the effort!

closed 2023-05-20 by Blueforcer

RobertoCravallo, 2023-12-22:
That is to sad, it would be really, really cool. Basically, this thing just rocks!!! Thank you, by the way!!! The "problem" is with lower case letters like the "w", which are hardy recognizable. Maybe you can generally use a "cleaner" font.

On the AWTRIX 2 this was possible and I saw a video with a really nice "w". ;-) 

[image]

Blueforcer, 2023-12-22:
> On the AWTRIX 2 this was possible and I saw a video with a really nice "w". ;-) 
>

Awtrix 2 uses the same Font ;-)