Blueforcer/awtrix3 #58: Toggle underscores

blaxer, 2023-04-12:
# Feature Request
Sorry I don't know the proper name for the app underscore that are the placeholders / highlight for each app.

But it would be very handy to be able to remove or turn off the underscores under the apps. For my particular use case, I will be using the clock only for very long periods of time and the underscores are superfluous. Not to mention at night the underscore is still pretty bright because it's always white. If you can already do this I apologize I couldn't find it in the docs for the API/MQTT.

### Is your feature request related to a problem?
No

If so, then add a clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

### Describe the solution / feature you'd like

See above

### Describe alternatives you've considered

N/A

### Additional context

Blueforcer, 2023-04-12:
These "underscores" showing the day of the week :-) 7 days, the brightest line is the current day.
Currently there is no option to switch it off, since its part of the time and day app.

blaxer, 2023-04-14:
Ahh okay thank you for the enlightenment :) Maybe I will try to mimic the clock the best I can with a custom app. Cheers!

Blueforcer, 2023-04-14:
I don't say I will not inplement your request :)

closed 2023-04-15 by Blueforcer

blaxer, 2023-04-15:
Thank you for implementing this I really appreciate it!

commit 17c34343 (2023-04-15): 0.52

- Changes setting keys to keep data small, please update your API requests!
- Adds setting to disable weekdays. closes #58
- Adds setting to change weekday color
- Smaller bugfixes