Blueforcer/awtrix3 #59: [BUG] Crash after 1h without Internet

oschick, 2023-04-13:
# Bug report

## Describe the bug
The Clock crashes if its running without Internet access for over one hour.
Probably related to update check introduced in version 0.47

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.51
## To Reproduce
Disable internet access for the clock and let it run for over one hour

## Expected behavior
Clock should not crash and the update check should return error or no update available

## Screenshots
[image]

udo1toni, 2023-04-15:
Hm. As my Ulanzi never got internet Access I can't confirm that. Never had a problem with the clock not being able to get contact to the Server. Of course, NTP is set to my router, which provides an NTP service for my LAN.

oschick, 2023-04-15:
Just checked again, 
if i don't block Internet access for the clock it runs for 16h+ no problem, as soon as I block it in my Firewall it crashes after approx. 1 hour. 

That behavior is 100% consistent for me.

So it must be some kind of bug in a function that runs every hour.

NTP is of course also set to my local NTP server.

oschick, 2023-04-15:
After building my own firmware with disabled automatic update check, i can confirm that it was indeed an issue with the update check.
I was not able to find any issues with the process itself, so i added a setting to disable the automatic update check as a workaround.
I'm running some final tests and should have a pull request ready soon.
[image]

Console output from crash which occoured approx. 1min after update check failed:
```
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (loopTask)
Core  1 register dump:
PC      : 0x400920ff  PS      : 0x00060b36  A0      : 0x800914eb  A1      : 0x3ffb0310
A2      : 0x3ffbf788  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060b23  
A6      : 0x00060b23  A7      : 0x0000abab  A8      : 0xb33fffff  A9      : 0xffffffff
A10     : 0x3ffb3bfc  A11     : 0x3ffb6954  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x007bf788  A15     : 0x003fffff  SAR     : 0x0000000a  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4008a19c  LEND    : 0x4008a1a6  LCOUNT  : 0xffffffff

Backtrace: 0x400920fc:0x3ffb0310 0x400914e8:0x3ffb0350 0x4008f292:0x3ffb0370 0x4008f76e:0x3ffb0390 0x4008f8df:0x3ffb03d0 0x40085653:0x3ffb03f0 0x40085729:0x3ffb0420 0x400857b5:0x3ffb0440 0x40195851:0x3ffb0470 0x40195592:0x3ffb04e0 0x40191751:0x3ffb0500 0x400eb0c9:0x3ffb0520 0x400d5381:0x3ffb0540 0x400d5395:0x3ffb0580 0x400d5395:0x3ffb05c0 0x400d5395:0x3ffb0600 0x400d5395:0x3ffb0640 0x400d5395:0x3ffb0680 0x400d5395:0x3ffb06c0 0x400d5395:0x3ffb0700 0x400d5395:0x3ffb0740 0x400d5395:0x3ffb0780 0x400d5395:0x3ffb07c0 0x400d5395:0x3ffb0800 0x400d5395:0x3ffb0840 0x400d5395:0x3ffb0880 0x400d5395:0x3ffb08c0 0x400d5395:0x3ffb0900 0x400d5395:0x3ffb0940 0x400d5395:0x3ffb0980 0x400d5395:0x3ffb09c0 0x400d5395:0x3ffb0a00 0x400d5395:0x3ffb0a40 0x400d5395:0x3ffb0a80 0x400d5395:0x3ffb0ac0 0x400d5395:0x3ffb0b00 0x400d5395:0x3ffb0b40 0x400d5395:0x3ffb0b80 0x400d5395:0x3ffb0bc0 0x400d5395:0x3ffb0c00 0x400d5395:0x3ffb0c40 0x400d5395:0x3ffb0c80 0x400d5395:0x3ffb0cc0 0x400d5395:0x3ffb0d00 0x400d5395:0x3ffb0d40 0x400d5395:0x3ffb0d80 0x400d5395:0x3ffb0dc0 0x400d5395:0x3ffb0e00 0x400d5395:0x3ffb0e40 0x400d5395:0x3ffb0e80 0x400d5395:0x3ffb0ec0 0x400d5395:0x3ffb0f00 0x400d5395:0x3ffb0f40 0x400d5395:0x3ffb0f80 0x400d5395:0x3ffb0fc0 0x400d5395:0x3ffb1000 0x400d5395:0x3ffb1040 0x400d5395:0x3ffb1080 0x400d5395:0x3ffb10c0 0x400d5395:0x3ffb1100 0x400d5395:0x3ffb1140 0x400d5395:0x3ffb1180 0x400d5395:0x3ffb11c0 0x400d5395:0x3ffb1200 0x400d5395:0x3ffb1240 0x400d5395:0x3ffb1280 0x400d5395:0x3ffb12c0 0x400d5395:0x3ffb1300 0x400d5395:0x3ffb1340 0x400d5395:0x3ffb1380 0x400d5395:0x3ffb13c0 0x400d5395:0x3ffb1400 0x400d5395:0x3ffb1440 0x400d5395:0x3ffb1480 0x400d5395:0x3ffb14c0 0x400d5395:0x3ffb1500 0x400d5395:0x3ffb1540 0x400d5395:0x3ffb1580 0x400d5395:0x3ffb15c0 0x400d5395:0x3ffb1600 0x400d5395:0x3ffb1640 0x400d5395:0x3ffb1680 0x400d5395:0x3ffb16c0 0x400d5395:0x3ffb1700 0x400d5395:0x3ffb1740 0x400d5395:0x3ffb1780 0x400d5395:0x3ffb17c0 0x400d5395:0x3ffb1800 0x400d5395:0x3ffb1840 0x400d5395:0x3ffb1880 0x400d5395:0x3ffb18c0 0x400d5395:0x3ffb1900 0x400d5395:0x3ffb1940 0x400d5395:0x3ffb1980 0x400d5395:0x3ffb19c0 0x400d5395:0x3ffb1a00 0x400d5395:0x3ffb1a40 0x400d5395:0x3ffb1a80 0x400d5395:0x3ffb1ac0 0x400d5395:0x3ffb1b00 0x400d5395:0x3ffb1b40 |<-CONTINUES

ELF file SHA256: 762eb5c386e1ff78

E (7520) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
E (577) esp_core_dump_flash: No core du�␆�partition found!
E (577) esp_core_dump_flash: No core dump partition found!
[  2092][E][vfsConfiguration loaded
Framebuffer_GFX::begin Width: 32 Height: 8 Num Pixels: 256
```

Blueforcer, 2023-04-15:
please to not touch the webinterface, this would cause everyone to delete the config.json, because its getting messed up when you add a new option in the middle. 
im already work on this issue.

oschick, 2023-04-15:
Thanks for looking into the issue.
I noticed the broken config.json as well, but for me it was the logical place to put the option.
If you find a solution i still think having the option to disable the update check is useful, maybe in a location that does not break the config.

Blueforcer, 2023-04-15:
Currently i try to reproduce this error. I turned off the internet connection for the clock and set the update interval to 60s.
Unfortunally i only get
```
checkUpdate: Check firmwareversion  
[134953][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -1  
checkUpdate: Error in downloading version file
```
each minute, without any crash. Are you sure youre not sending anything else to the clock wich cause this crash?

oschick, 2023-04-15:
No, i just send my normal messages which work fine, as its running fine for 16+h with internet access.

I think i also got the same message and it crashed like 1 min after that, let me check again.

Blueforcer, 2023-04-15:
please try this version. it has debug messages enabled. Updateinterval also set to 60s

[firmware.zip](https://github.com/Blueforcer/awtrix-light/files/11240866/firmware.zip)

please post the messages around the crash

Blueforcer, 2023-04-15:
im a bit confused. because the updatecheck runs directly after boot the first time. In theory it should also crash there

oschick, 2023-04-15:
> please try this version. it has debug messages enabled. Updateinterval also set to 60s
> 
> [firmware.zip](https://github.com/Blueforcer/awtrix-light/files/11240866/firmware.zip)
> 
> please post the messages around the crash

Testing now

oschick, 2023-04-15:
Crashed again, but only after the 5-6th update check.

Same error as before:

```
onMqttMessage: MQTT message received at topic smarthome/awtrix/ole/custom/alarm
onMqttMessage: Payload:
setCurrentApp: Publish current app grid
Check Update
checkUpdate: Check firmwareversion..
[372936][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -1
checkUpdate: Error downloading version file
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (loopTask)
Core  1 register dump:
PC      : 0x4008ee63  PS      : 0x00060d36  A0      : 0x4008ec66  A1      : 0x3ffb02c0
A2      : 0x00000000  A3      : 0xffffffff  A4      : 0x00060d20  A5      : 0x00060b23  
A6      : 0x007b4d44  A7      : 0x003fffff  A8      : 0x800914f7  A9      : 0x3ffb0350
A10     : 0x3ffcf874  A11     : 0x3ffcf874  A12     : 0x400852fa  A13     : 0x00060d23  
A14     : 0x00060d23  A15     : 0x00000001  SAR     : 0x0000000a  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x4008a19c  LEND    : 0x4008a1a6  LCOUNT  : 0xffffffff  

Backtrace: 0x4008ee60:0x3ffb02c0 0x4008ec63:0x3ffb0390 0x4008f795:0x3ffb03b0 0x4008f8df:0x3ffb03f0 0x40085653:0x3ffb0410 0x40085729:0x3ffb0440 0x400857b5:0x3ffb0460 0x40196591:0x3ffb0490 0x40196609:0x3ffb04d0 0x401927af:0x3ffb04f0 0x400ec021:0x3ffb0510 0x400d4baa:0x3ffb0540 0x400d5214:0x3ffb0560 0x400d547f:0x3ffb0580 0x400d5489:0x3ffb05c0 0x400d5489:0x3ffb0600 0x400d5489:0x3ffb0640 0x400d5489:0x3ffb0680 0x400d5489:0x3ffb06c0 0x400d5489:0x3ffb0700 0x400d5489:0x3ffb0740 0x400d5489:0x3ffb0780 0x400d5489:0x3ffb07c0 0x400d5489:0x3ffb0800 0x400d5489:0x3ffb0840 0x400d5489:0x3ffb0880 0x400d5489:0x3ffb08c0 0x400d5489:0x3ffb0900 0x400d5489:0x3ffb0940 0x400d5489:0x3ffb0980 0x400d5489:0x3ffb09c0 0x400d5489:0x3ffb0a00 0x400d5489:0x3ffb0a40 0x400d5489:0x3ffb0a80 0x400d5489:0x3ffb0ac0 0x400d5489:0x3ffb0b00 0x400d5489:0x3ffb0b40 0x400d5489:0x3ffb0b80 0x400d5489:0x3ffb0bc0 0x400d5489:0x3ffb0c00 0x400d5489:0x3ffb0c40 0x400d5489:0x3ffb0c80 0x400d5489:0x3ffb0cc0 0x400d5489:0x3ffb0d00 0x400d5489:0x3ffb0d40 0x400d5489:0x3ffb0d80 0x400d5489:0x3ffb0dc0 0x400d5489:0x3ffb0e00 0x400d5489:0x3ffb0e40 0x400d5489:0x3ffb0e80 0x400d5489:0x3ffb0ec0 0x400d5489:0x3ffb0f00 0x400d5489:0x3ffb0f40 0x400d5489:0x3ffb0f80 0x400d5489:0x3ffb0fc0 0x400d5489:0x3ffb1000 0x400d5489:0x3ffb1040 0x400d5489:0x3ffb1080 0x400d5489:0x3ffb10c0 0x400d5489:0x3ffb1100 0x400d5489:0x3ffb1140 0x400d5489:0x3ffb1180 0x400d5489:0x3ffb11c0 0x400d5489:0x3ffb1200 0x400d5489:0x3ffb1240 0x400d5489:0x3ffb1280 0x400d5489:0x3ffb12c0 0x400d5489:0x3ffb1300 0x400d5489:0x3ffb1340 0x400d5489:0x3ffb1380 0x400d5489:0x3ffb13c0 0x400d5489:0x3ffb1400 0x400d5489:0x3ffb1440 0x400d5489:0x3ffb1480 0x400d5489:0x3ffb14c0 0x400d5489:0x3ffb1500 0x400d5489:0x3ffb1540 0x400d5489:0x3ffb1580 0x400d5489:0x3ffb15c0 0x400d5489:0x3ffb1600 0x400d5489:0x3ffb1640 0x400d5489:0x3ffb1680 0x400d5489:0x3ffb16c0 0x400d5489:0x3ffb1700 0x400d5489:0x3ffb1740 0x400d5489:0x3ffb1780 0x400d5489:0x3ffb17c0 0x400d5489:0x3ffb1800 0x400d5489:0x3ffb1840 0x400d5489:0x3ffb1880 0x400d5489:0x3ffb18c0 0x400d5489:0x3ffb1900 0x400d5489:0x3ffb1940 0x400d5489:0x3ffb1980 0x400d5489:0x3ffb19c0 0x400d5489:0x3ffb1a00 0x400d5489:0x3ffb1a40 0x400d5489:0x3ffb1a80 0x400d5489:0x3ffb1ac0 0x400d5489:0x3ffb1b00 0x400d5489:0x3ffb1b40 |<-CONTINUES

ELF file SHA256: 36a1bed184418802

E (5667) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
E (579) esp_core_dump_flash: No core du�␀�partition found!
E (579) esp_core_dump_flash: No core dump partition found!
```

Blueforcer, 2023-04-15:
mine runs fine after 40 download tries.

in the log i can see that the crash happens **after** the download try

i now changed the WiFiClientSecure object from dynamic to static wich should free the memory when the functions exit. If this also doesnt work, i need to dig deeper into it tomorrow

[firmware.zip](https://github.com/Blueforcer/awtrix-light/files/11240897/firmware.zip)

oschick, 2023-04-15:
I am now testing without sending any custom apps, just stock firmware and so far it has not crashed.
Which does not make sense, since i am sending the same messages with or without internet and they work fine.
Maybe some sort of issue relating to shared memory idk.

> mine runs fine after 40 download tries.
>
> in the log i can see that the crash happens **after** the download try
>
> i now changed the WiFiClientSecure object from dynamic to static wich should free the [...]

Gonna install it now and let it run overnight. hopefully i'll have a update soon.

Thank you again for looking into this weird issue.

Blueforcer, 2023-04-15:
Is it possible that the message frequency is too high at some point? I remember one user wich tells me about this problem. But we both can't reproduce it anymore 😅

oschick, 2023-04-16:
Have you set the check freq. back to 1h in the latest version, i dont see any update messages in the console.

> Is it possible that the message frequency is too high at some point? I remember one user wich tells me about this problem. But we both can't reproduce it anymore 😅

I dont think so, i am sending 5 custom apps every 20 seconds, the content also does not change in size, so i dont think that should be an issue, that also wouldn't explain why it is suddenly working fine with internet access, while consistently crashing without.

oschick, 2023-04-16:
Sadly not fixed in the latest version you send:
```
setCurrentApp: Publish current app temp_in
Check Update
checkUpdate: Check firmwareversion
[3611933][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -1
checkUpdate: Error in downloading version file
setCurrentApp: Publish current app Time
onMqttMessage: MQTT message received at topic smarthome/awtrix/ole/custom/grid
onMqttMessage: Payload:  { "pos": 3,  "text": "679W, 2kWh",   "color": "ffff00", "duration": "15",  "icon" :"21256" }
onMqttMessage: MQTT message received at topic smarthome/awtrix/ole/custom/solar
onMqttMessage: Payload:  { "pos": 4,  "text": "0W",   "color": "ffff00", "duration": "15",  "icon" :"40589" }
onMqttMessage: MQTT message received at topic smarthome/awtrix/ole/custom/temp_out
onMqttMessage: Payload:  { "pos": 5,  "text": "8°C",   "color": "ffff00", "duration": "10",  "icon" :"4285" }
onMqttMessage: MQTT message received at topic smarthome/awtrix/ole/custom/temp_in
onMqttMessage: Payload:  { "pos": 6,  "text": "19.7°C",   "color": "90EE90", "duration": "10",  "icon" :"4286" }
onMqttMessage: MQTT message received at topic smarthome/awtrix/ole/custom/alarm
onMqttMessage: Payload:
setCurrentApp: Publish current app Date
setCurrentApp: Publish current app grid
onMqttMessage: MQTT message received at topic smarthome/awtrix/ole/custom/grid
onMqttMessage: Payload:  { "pos": 3,  "text": "675W, 2kWh",   "color": "ffff00", "duration": "15",  "icon" :"21256" }
onMqttMessage: MQTT message received at topic smarthome/awtrix/ole/custom/solar
onMqttMessage: Payload:  { "pos": 4,  "text": "0W",   "color": "ffff00", "duration": "15",  "icon" :"40589" }
onMqttMessage: MQTT message received at topic smarthome/awtrix/ole/custom/temp_out
onMqttMessage: Payload:  { "pos": 5,  "text": "8°C",   "color": "ffff00", "duration": "10",  "icon" :"4285" }
onMqttMessage: MQTT message received at topic smarthome/awtrix/ole/custom/temp_in
onMqttMessage: Payload:  { "pos": 6,  "text": "19.7°C",   "color": "90EE90", "duration": "10",  "icon" :"4286" }
onMqttMessage: MQTT message received at topic smarthome/awtrix/ole/custom/alarm
onMqttMessage: Payload:
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
Core  1 register dump:
PC      : 0x400920ff  PS      : 0x00060b36  A0      : 0x800914eb  A1      : 0x3ffb0310
A2      : 0x3ffbf788  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060b23  
A6      : 0x00060b23  A7      : 0x0000abab  A8      : 0xb33fffff  A9      : 0xffffffff
A10     : 0x3ffb4c04  A11     : 0x3ffb6954  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x007bf788  A15     : 0x003fffff  SAR     : 0x0000000a  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x4008a19c  LEND    : 0x4008a1a6  LCOUNT  : 0xffffffff  

Backtrace: 0x400920fc:0x3ffb0310 0x400914e8:0x3ffb0350 0x4008f292:0x3ffb0370 0x4008f76e:0x3ffb0390 0x4008f8df:0x3ffb03d0 0x40085653:0x3ffb03f0 0x40085729:0x3ffb0420 0x400857b5:0x3ffb0440 0x4019693d:0x3ffb0470 0x4019667e:0x3ffb04e0 0x4019283d:0x3ffb0500 0x400ec0dd:0x3ffb0520 0x400d5489:0x3ffb0540 0x400d549d:0x3ffb0580 0x400d549d:0x3ffb05c0 0x400d549d:0x3ffb0600 0x400d549d:0x3ffb0640 0x400d549d:0x3ffb0680 0x400d549d:0x3ffb06c0 0x400d549d:0x3ffb0700 0x400d549d:0x3ffb0740 0x400d549d:0x3ffb0780 0x400d549d:0x3ffb07c0 0x400d549d:0x3ffb0800 0x400d549d:0x3ffb0840 0x400d549d:0x3ffb0880 0x400d549d:0x3ffb08c0 0x400d549d:0x3ffb0900 0x400d549d:0x3ffb0940 0x400d549d:0x3ffb0980 0x400d549d:0x3ffb09c0 0x400d549d:0x3ffb0a00 0x400d549d:0x3ffb0a40 0x400d549d:0x3ffb0a80 0x400d549d:0x3ffb0ac0 0x400d549d:0x3ffb0b00 0x400d549d:0x3ffb0b40 0x400d549d:0x3ffb0b80 0x400d549d:0x3ffb0bc0 0x400d549d:0x3ffb0c00 0x400d549d:0x3ffb0c40 0x400d549d:0x3ffb0c80 0x400d549d:0x3ffb0cc0 0x400d549d:0x3ffb0d00 0x400d549d:0x3ffb0d40 0x400d549d:0x3ffb0d80 0x400d549d:0x3ffb0dc0 0x400d549d:0x3ffb0e00 0x400d549d:0x3ffb0e40 0x400d549d:0x3ffb0e80 0x400d549d:0x3ffb0ec0 0x400d549d:0x3ffb0f00 0x400d549d:0x3ffb0f40 0x400d549d:0x3ffb0f80 0x400d549d:0x3ffb0fc0 0x400d549d:0x3ffb1000 0x400d549d:0x3ffb1040 0x400d549d:0x3ffb1080 0x400d549d:0x3ffb10c0 0x400d549d:0x3ffb1100 0x400d549d:0x3ffb1140 0x400d549d:0x3ffb1180 0x400d549d:0x3ffb11c0 0x400d549d:0x3ffb1200 0x400d549d:0x3ffb1240 0x400d549d:0x3ffb1280 0x400d549d:0x3ffb12c0 0x400d549d:0x3ffb1300 0x400d549d:0x3ffb1340 0x400d549d:0x3ffb1380 0x400d549d:0x3ffb13c0 0x400d549d:0x3ffb1400 0x400d549d:0x3ffb1440 0x400d549d:0x3ffb1480 0x400d549d:0x3ffb14c0 0x400d549d:0x3ffb1500 0x400d549d:0x3ffb1540 0x400d549d:0x3ffb1580 0x400d549d:0x3ffb15c0 0x400d549d:0x3ffb1600 0x400d549d:0x3ffb1640 0x400d549d:0x3ffb1680 0x400d549d:0x3ffb16c0 0x400d549d:0x3ffb1700 0x400d549d:0x3ffb1740 0x400d549d:0x3ffb1780 0x400d549d:0x3ffb17c0 0x400d549d:0x3ffb1800 0x400d549d:0x3ffb1840 0x400d549d:0x3ffb1880 0x400d549d:0x3ffb18c0 0x400d549d:0x3ffb1900 0x400d549d:0x3ffb1940 0x400d549d:0x3ffb1980 0x400d549d:0x3ffb19c0 0x400d549d:0x3ffb1a00 0x400d549d:0x3ffb1a40 0x400d549d:0x3ffb1a80 0x400d549d:0x3ffb1ac0 0x400d549d:0x3ffb1b00 0x400d549d:0x3ffb1b40 |<-CONTINUES

ELF file SHA256: c2849c85d60fd8fe

E (11081) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
E (579) esp_core_dump_flash: No core du�␀�partition found!
E (579) esp_core_dump_flash: No core dump partition found!
```

Blueforcer, 2023-04-16:
But now I'm pretty sure it has nothing to do with the update.

oschick, 2023-04-16:
Tonight it crashed consistently every hour even with internet access, so you seem to be correct.

I'm gonna test some more combinations today and see if i can pinpoint exactly what causes it.

oschick, 2023-04-16:
After some tests:

No Internet:
MQTT Apps & Update Check: Crash
MQTT Apps (only send once, no reg. updates) & Update Check: Crash

No MQTT Apps & Update Check: No Crash
MQTT Apps & **No** Update Check: No Crash

All tests were using the v. 0.54 code base. So still the old WifiClientSecure.

Blueforcer, 2023-04-16:
After some searching on the net, it could be a mix of both. using a ssl https request cost huge ammount of stack memory.
if you also sending mqtt messages, the stack could overflow.
Right now, im searching for a solution to increase the stack size. And i will create a dev setting to disable (or enable) update searching

commit 0563ea89 (2023-04-16): V0.55

- fix color correction and temperature in settings API
- deactivates auto update search and following crashes (can be activated in dev.json) closes #59
- fixes colored fragments in customapps
- adds Linechart to notify and customapps
- adds option to disable autoscaling for barchart and linechart. closes #64
commit 6f6274a5 (2023-04-16): V0.55

- adds doc link to webinterface
- adds debug outputs to serial
- fix color correction and temperature in settings API
- deactivates auto update search and following crashes (can be activated in dev.json) closes #59
- fixes colored fragments in customapps
- adds Linechart to notify and customapps
- adds option to disable autoscaling for barchart and linechart. closes #64
closed 2023-04-16 by Blueforcer