Blueforcer/awtrix3 #67: [BUG]

pdeheij, 2023-04-17:
# Bug report

Clock is bricked 
## Additional information
Clock stays on 02:36:34 and wil not reset with button, also the com port does work anymore, flashing not posible
- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version:  v0.54 ]

## To Reproduce
Steps to reproduce the behavior:
Can not reproduce, clock is bricked

## Expected behavior
A clear and concise description of what you expected to happen.

## Screenshots
If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context
Add any other context about the problem here.

Blueforcer, 2023-04-17:
Clock isn't brickable.
Can you turn it off while holding left and right button simultaneously?

Nash2112, 2023-04-17:
Disable MQTT. V0.53 OK  V.055 Start, Crash, Start, etc

Nash2112, 2023-04-17:
Die Uhr startet nach dem Update, zeigt IP Adresse an, dann kommt ein noch oben fliegendes Z und die Uhr startet neu. 
Uhr Ulanzi, Version 0.55, Iobroker MQTT Server

The clock starts after the update, shows IP address, then comes a still flying Z above and the clock restarts.
Clock Ulanzi, version 0.55, Iobroker MQTT Server

Blueforcer, 2023-04-17:
> Die Uhr startet nach dem Update, zeigt IP Adresse an, dann kommt ein noch oben fliegendes Z und die Uhr startet neu.
> Uhr Ulanzi, Version 0.55, Iobroker MQTT Server
>
> The clock starts after the [...]

Dann mach doch dafür ein eigenes issue auf? Oder komm auf den discord.

pdeheij, 2023-04-17:
> Clock isn't brickable by software. Can you turn it off while holding left and right button simultaneously?

Yes it turns off, but does not turn on anymore, still no comport

Blueforcer, 2023-04-17:
Is the battery charged? It will not turn on with empty battery even plugged in.

There are some reports of dead clocks. But this is a hardware failure and has nothing to do with this software. Contact ulanzi for a replacement.

closed 2023-04-17 by Blueforcer

martin131, 2023-04-21:
@pdeheij Any solution, please? Battery seems to be charged but i cannot turn on clock

Blueforcer, 2023-04-21:
If the com port doesn't work it's a hardware failure and has nothing to do with the firmware. Contact Ulanzi to get a replacement

pdeheij, 2023-04-21:
@martin131 No, contacted Ulanzi, waiting for answer, hoping in replacement.

Blueforcer, 2023-04-21:
> @Blueforcer  It is very strange. Yesterday my clocks were delivered. I flashed Awtrix Light on both. And today both of them stopped working. Both was on different USB adapter. 

There are many users the last time, who have broken clocks. But this seems to depend on the latest batches. There are lots of older users without problems.

The ESP has no physical connection to the power management. It only gets the 3,3V if you turn it on. That's it. If you doesn't hear a short beep while turning on, the ESP gets no power.

Blueforcer, 2023-04-21:
heres a photo from a user with the same problem
Its clearly a hardware failure
[image]

martin131, 2023-04-21:
OK, I send an email to Ulanzi support. I bought three clocks total so I will try to turn on the third one.

Blueforcer, 2023-04-21:
You can also try to let it run with the original firmware for some days

martin131, 2023-04-21:
Yes, I will. I turned on "matrix" animation and full brightness 🙂

martin131, 2023-04-21:
I think this is weird. Temperature sensor was very hot while charging. Same behavior on both clocks. 
[image]

Blueforcer, 2023-04-21:
My device is currently connected to my PC and receiving a low amount of current, but enough to prevent discharge. Therefore, I haven't experienced any issues for over two months. However, it appears that the charging regulator may be undersized, which would explain the blown ICs and high temperature.

MrRodz, 2023-04-22:
[pdeheij](https://github.com/pdeheij)

> > Clock isn't brickable by software. Can you turn it off while holding left and right button simultaneously?
> 
> Yes it turns off, but does not turn on anymore, still no comport

Hi I have the same issue, when you plug the charger in does the light on the back light up for you?

supremelucifer, 2023-04-22:
> [pdeheij](https://github.com/pdeheij)
>
>
> Hi I have the same issue, when you plug the charger in does the light on the back light up for you?

Got the same thing and mine does light on the back

Dawnes, 2023-04-22:
Flashed 0.57 over 0.56 that had been running a week or so, and 1 hour later I had exactly this issue. Completely dead. No serial. Charging light is still working.  Not saying it is firmware but it happened right after updating.

supremelucifer, 2023-04-23:
Just send them a mail. They will send you a new one without wanting the old one back. 
[image]

djungelola, 2023-04-24:
> I have the same thing. During the night the clock disconnected from the network (went unavailable in HA) and I couldn't switch on the clock. The power led is on, and after a couple of hours it is turning green. Can't get is powered up again though, no beep no display.
> 
> I am wondering how to open the case. I saw the photo but I didn't succeed yet opening the case without destroying it.

I know how to do it :-) Did it yesterday myself.
The screen is glued to the case. I used a hairdryer to warm it up for about 10 min. The gently stick in something between the screen and the case. Warm more if needed. Then remove the screws and you have access to the circuitboard.

i didnt find any components that looked burned at all, but still only the charging led is lit up. seems to charge the battery though.
very strange that so many people have this problem to be honest.

Br Ola
[image]

floschibo, 2023-04-24:
Same here. No way to get the clock running again. Clock was charging for three days with matrix off. Contacted support now.

Dawnes, 2023-04-24:
@supremelucifer I got a refund too, so all is good. It must be a faulty batch.

pdeheij, 2023-04-25:
I also get a new one, but had a good advise form Ulanzi. The charcher i use was a Samsung with the capabilty for fastcharge. This means it is posible that 9V is provided to the clock. Use a old 5V charger !! was the advice. Must say that Ulanzi has a very good service!. Next week i have a few days off and try to open the defective clock. Maybe somthing interesting comes out.

martin131, 2023-04-25:
I can confirm that one of my clocks died at Quick Charge 3.0 charger. About the second one, I am not sure.

Mauricevb, 2023-04-25:
I was using an old 5V 1A iPhone charger. So I don't think it has anything to do with the charger 🤔

MrRodz, 2023-04-25:
They are sending me a new one. They say it should be a 5v1a charger. I was using a 1.2a charger.

MrRodz, 2023-04-25:
[image]

so I’ve pulled my clock apart and traced my fault to this regulator. I have 3.7v going in and nothing coming out. It has GU2Z written on it. I can’t find that one online but have seen some others but not sure if they would be compatible.

Blueforcer, 2023-04-25:
It's everytime the same IC. I posted a photo some posts before where it blows up. I think this is kind of switch, transistor whatever wich gives the power to the esp when you turn it on.

Dawnes, 2023-04-25:
Question now is, will a replacement unit be any better? Was it bad luck for this batch, or did they change the circuit/components recently and all will fail?

@Blueforcer Your clock is probably older and does not have this issue. Any idea if the circuitboard is different?
@MrRodz Do you think a bypass of this chip would be an option? I really can't tell what it is supposed to do, but if it is a regulator and 3.7 is coming in, we should be ok right?

MrRodz, 2023-04-25:
No a bypass will potentially blow the esp, from what data sheets i found it gives a constant 3,3v out to the esp from a 2.4-5.5v input. The supply comes through a mosfet too witch i think turns off the supply if the voltage goes too low from the battery. 

Hopefully i can just replace the regulator if i can find one. 

There is no guarantee that the new ones will be any better, guess we will just have to find out. 

I did find 

https://twitter.com/rm83/status/1621194459965788160

With some pics of a working one from two months ago, but the board looks the same

Blueforcer, 2023-04-25:
In my discord server (https://discord.gg/cyBCpdx), all users with the board date 2023-3-2 have this problem.
[image]

I received my two clocks before this date. I will check my pcb stamp tommorow

MrRodz, 2023-04-25:
Unfortunately thats the board i have. Did any of your other users find a fix or work around?

Blueforcer, 2023-04-28:
> Unfortunately thats the board i have. Did any of your other users find a fix or work around?

https://discord.com/channels/546407049148366859/1096822468737511574/1101222937501126778

barbiudo, 2023-05-06:
same issue here, clock dead, board date 2023-3-2  :-(

StormyKnight17, 2023-05-19:
So my first clock died within 6hrs of use. Second is going well so far, running for 24hrs so far ....

darkpoet78, 2023-06-09:
My first still runs.  My second suddenly froze up while displaying the time.  I could turn it off but not on again... I'll contact them direct on Aliexpress (a good reason to always buy from the official store even if it's a few bucks more) but in the meantime... I took some good pics while pulling it apart if anyone else feels the need to pull theirs apart without destroying it.

[Imgur Link](https://imgur.com/a/2p2miqF)

scarolan, 2023-06-11:
Same failure here, clock froze, after reset screen won't come on at all.  I will contact Ulanzi about the HW failure.

scarolan, 2023-06-11:
> Just send them a mail. They will send you a new one without wanting the old one back. [image]

Shame to waste a good device. Has anyone had luck de-soldering and replacing the failed voltage regulator?

ptrkp, 2023-06-14:
Today I received a new clock after the first one died after a month or so.
Did take the broken one apart, would be nice if it is fixable.

ptrkp, 2023-06-15:
Removed the faulty charge controller and the battery and solder a 5v to 3.3v stepdown converter on to the pcb.
Clock working again.

[image]

Mauricevb, 2023-06-16:
> Removed the faulty charge controller and the battery and solder a 5v to 3.3v stepdown converter on to the pcb. Clock is working again.

Nice! Would you mind sharing the exact part(s) that you used?

darkpoet78, 2023-06-16:
Yes, please do... over on Discord someone recommended a STI3408B.  I've got a few on the way from Aliexpress.  I'll also report back if it was a success or not.

ptrkp, 2023-06-16:
I'm not responsible if somethings goes wrong, you need to have knowledge of how to solder and how to work with a multimeter.
Remove the charge controller and clean the surface. Check with the multimeter if there are no shortages anymore.
Check if there is 5v.

Solder the 3.3v step down converter with some wires to the pcb

Step down 3.3 converter
https://www.amazon.com/Alinan-Efficiency-Converter-Regulator-Stabilized/dp/B09W8P1QNM or something similar.

[image]

[image]

stefangries, 2023-06-23:
I can confirm, that replacing the part at position A_U1 (GU2B) with a new STI3408B fixes this problem. I got mine on [AliExpress](https://www.aliexpress.com/item/1005004307132691.html).

darkpoet78, 2023-06-23:
Still waiting on my replacement STI3408Bs but in the meantime, I contacted Ulanzi through their official store and they're charging less than a dollar for a replacement.  It's great to see a company like this stand behind their product even if they screwed something up.  Here's hoping they rectified the issue... has anyone opened up one from the new batch to confirm it doesn't use the GU2B any more?

trip5, 2023-06-27:
@stefangries Yup.  SMD soldering isn't my thing.  After removing the old regulator, I made 3 attempts but just ended up ripping the trace on the board.  I had an adjustable voltage regulator laying around so I set it to 3.3V and followed the pics @ptrkp posted.  Thanks a lot!  It might be lacking the battery but it works... although lol a bit weird that you have to press the left and right buttons to turn it on after plugging it in... but it works.

Calimerorulez, 2023-06-29:
Mine just died too, bought late April on AliExpress. It froze on displaying the time. I've sent them a message.

ptrkp, 2023-06-29:
> @stefangries Yup. SMD soldering isn't my thing. After removing the old regulator, I made 3 attempts but just ended up ripping the trace on the board. I had an adjustable voltage regulator laying [...]

Are your temperature and humidity readings still working? Mine are dead not sure what happened there.

trip5, 2023-06-30:
My temp and humidity sensor still function, yes but... not more accurate than before.  So, still basically useless.  If you accidentally hit them with 5V, you could have killed them.  ESPs can survive that voltage but sensors can be fried.  Maybe static, too...  I've also killed BME sensors with heat (ie too much soldering) so there's a million ways you could have killed them.  

Honestly I'd consider it a win to have the board still working... I was willing to throw the whole board out and try to wire up a new ESP32 to the buttons and screen.

stefangries, 2023-06-30:
> Are your temperature and humidity readings still working? Mine are dead not sure what happened there.

Mine are still working. However, their values are completely unusable, as they are much too high. However, they were also not correct before. I think it is simply warmer in the housing where the sensor is located.

ptrkp, 2023-06-30:
Try to hook an external one SHT31 but that also did not work, so I think more things got fried when the clock went dead.
But the values where useless so I can live without it.

Mauricevb, 2023-06-30:
> I'm not responsible if somethings goes wrong, you need to have knowledge of how to solder and how to work with a multimeter. Remove battery, you don't need that anymore, clock will only work via usb-c. Remove the charge controller and clean the surface. Check with the multimeter if there are no shortages anymore. Check if there is 5v.
> 
> Solder the 3.3v step down converter with some wires to the pcb
> 
> Step down 3.3 converter https://www.amazon.com/Alinan-Efficiency-Converter-Regulator-Stabilized/dp/B09W8P1QNM or something similar.
> ....
> Be care full, I broke the reset button upper left near usb-c connector.

Thank you @ptrkp! I was able to fix mine with this. It was hard to remove the charge controller, but I managed do it apparently.

Calimerorulez, 2023-06-30:
I soldered a step down converter like @ptrkp did, and the clock is working again. Thank you for the pictures 😃

scarolan, 2023-07-13:
@ptrkp you are the real MVP...thanks for the photo and tutorial

JoFranke, 2023-08-01:
Excellent, thanks for all that information. I actually have three bricked devices now, and I hope Ulanzi will replace them as they are just a few weeks old.

The question is, did they fix the issue, or will the replacement ones fail again - does anyone has any experiance with the replacement? If it is not fixed, I will try to get a refund instead....thanks.

Blueforcer, 2023-08-01:
> The question is, did they fix the issue, or will the replacement ones fail again - does anyone have any experiance with the replacement? If it is not fixed, I will try to get a refund instead....thanks.

according to the productmanager, its fixed.

JoFranke, 2023-08-01:
> according to the productmanager, its fixed.

Thats good news, thanks for the info - I'm from Germany and could send you two bad ones if they dont want them back. PM for details :)

Blueforcer, 2023-08-01:
How to PM in github? 😮

JoFranke, 2023-08-01:
> How to PM in github? 😮

Sent you a mail to your website admin address

[4 more replies]