Blueforcer/awtrix3 #68: [BUG]

Nash2112, 2023-04-17:
Die Uhr startet nach dem Update, zeigt IP Adresse an, dann kommt ein noch oben fliegendes Z und die Uhr startet neu. Uhr Ulanzi, Version 0.55, Iobroker MQTT Server
Wenn ich den MQTT Server auf IOBroker ausschalten, funktioniert die Uhr

The clock starts after the update, shows IP address, then comes a still flying Z above and the clock restarts.
Clock Ulanzi, version 0.55, Iobroker MQTT Server
When I turn off the MQTT server on IOBroker, the clock works

## Zusätzliche Informationen  - Beteiligte Geräte: - Modell: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001) - AWTRIX-LIGHT-Version: 0.55

Blueforcer, 2023-04-17:
Klingt danach als hättest du einige mqtt Messages auf retained die beim Connect der Uhr direkt los gesendet werden.
Das "Z" zeigt an das die Uhr per api ausgeschaltet wurde. Von alleine passiert das nicht.

Nash2112, 2023-04-17:
Das Problem habe ich aber erst nach dem Update auf V 0.55. Ich schaue noch nach, ob der Fehler im Script liegt

Nash2112, 2023-04-17:
Habe bei MQTT-Server "Eigene States beim Verbinden publizieren" ausgeschaltet. Jetzt funktioniert es

closed 2023-04-17 by Blueforcer