Blueforcer/awtrix3 #69: Update my old awtrix lcd

diegomarprogramacion, 2023-04-17:
Would it be possible to upgrade my old lcd from awtrix with esp8266? or esp32?
Thanks!

Blueforcer, 2023-04-17:
You mean Awtrix 2 ?

Yes its fully compatible with the awtrix_upgrade enviroment thanks to @Elfish .
https://github.com/Blueforcer/awtrix-light/blob/main/platformio.ini#L29
Its made for ESP32 D1 mini for 1:1 replacement
[image]

For now you have to compile it for your own, it will get its own flasher soon

closed 2023-04-17 by Blueforcer

Blueforcer, 2023-04-30:
Its now added in every release, including online flasher and OTA updates
https://github.com/Blueforcer/awtrix-light/releases/tag/0.60

Elektroniker0486, 2023-06-17:
Hallo, 
ich habe mir jetzt den ESP32 mini besorgt und AWTRIX Light erfolgreich geflasht.
Der AP Mode war sichtbar und ich hab den ESP in mein Netzwerk /WLAN eingebunden. Wenn ich den ESP32 mini nun in die AWTRIX2 einbaue/aufstecke dann zeigt mir die Matrix keine sinnvolle Zeichen. Eher ein Wirrwarr aus bunten LED Pixel. Im Webif gibt es ja keine Einstellungen zur Matrix. Wie kann ich das Problem lösen damit er alles richtig anzeigt?

Vielen Dank im Vorraus!

Beste Grüße

Blueforcer, 2023-06-17:
Matrix Layout ändern.
https://blueforcer.github.io/awtrix-light/#/dev

Torrid16V, 2024-06-23:
> Matrix Layout ändern. https://blueforcer.github.io/awtrix-light/#/dev

Kannst du den Link nochmals teilen?
Ich würde gerne auch meine Awtrix 2 updaten und habe nun das gleich Problem.

Blueforcer, 2024-06-23:
> > Matrix Layout ändern. https://blueforcer.github.io/awtrix-light/#/dev
> 
> Kannst du den Link nochmals teilen?
> Ich würde gerne auch meine Awtrix 2 updaten und habe nun das gleich Problem.

https://blueforcer.github.io/awtrix3/#/dev

DennisXK, 2025-07-30:
Hallo, habe meine Awtrix 2 auch auf die Awtrix 3 upgedatet.  Verbaut habe einen ESP32 mit 30pins. Soweit so gut, läuft auch mit Home Assistans (customs apps). Mir ist aufgefallen das diese Awtrix leider keinen Information(Sensoren) anzeigt. Zb. IP, free ram usw.

Bei den geflashten Ulanzi Display bekomme ich dise Sensordaten im Home Assistant angezeigt.