Blueforcer/awtrix3 #70: [BUG]

matze20211, 2023-04-19:
# Bug report

## Describe the bug

MQTT power on no longer working, after power off, if i send power on, the state changed to "on" but the Display stays off. I have to restart the device to work 

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.56/0.57 

## To Reproduce

Steps to reproduce the behavior:

MQTT power off, power on, every try

Blueforcer, 2023-04-19:
According to the changelog, the way how this API works has changed. Please have a look at the documentation.

closed 2023-04-19 by Blueforcer

matze20211, 2023-04-19:
and how? MQTT API Documentation: <h2 id="turn-display-on-or-off"><a href="https://blueforcer.github.io/awtrix-light/#/api?id=turn-display-on-or-off" data-id="turn-display-on-or-off" class="anchor"><span>Turn display on or off</span></a></h2>

Topic | URL | Payload/Body | HTTP method
-- | -- | -- | --
[PREFIX]/power | http://[IP]/api/power | true/false or 1/0 | POST

[Turn display on or off](https://blueforcer.github.io/awtrix-light/#/api?id=turn-display-on-or-off)
Topic 	URL 	Payload/Body 	HTTP method
[PREFIX]/power 	http://[IP]/api/power 	true/false or 1/0 	POST

is not working?