Blueforcer/awtrix3 #72: Temperature has no decimals in Home Assistant entity

wilfredktlr, 2023-04-20:
# Feature Request

### Is your feature request related to a problem?

No.

### Describe the solution / feature you'd like

Currently you can set the `temp_dec_places` in the file `dev.json` so the display shows the specified number of decimals. However the value of the Home Assistant entity doesn't contain any decimals.

### Describe alternatives you've considered

I have tried setting the precision of the entity in Home Assistant but that didn't work.

### Additional context

n/a

smarthomejunkie, 2023-04-22:
I wouldn't use the temperature entity of the clock. The value is way off. I think it's better to use a temperature sensor in the room and show that value on the clock.

wilfredktlr, 2023-04-22:
Agreed. The clock gets a bit warm and that distorts the value.

closed 2023-04-22 by wilfredktlr