Blueforcer/awtrix3 #75: My clock doesn't turn on anymore

Mauricevb, 2023-04-21:
# Question

My clock (which is on v0.57) randomly turned off today and I can't turn it on now. It was connected to the power supply and I tried turning it on by holding the arrow buttons for 3 (or more) seconds. I also tried resetting the device using the reset-button in the back, but that doesn't do anything. The light next to the charging port still lights up when I connect a power supply. 

Has anyone seen this issue? Is there another way to reset the clock? Unfortunately my computer does no longer recognise it.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.57

Blueforcer, 2023-04-21:
its a hardware issue
duplicate #67

closed 2023-04-21 by Blueforcer

Mauricevb, 2023-04-21:
Oops, how could I have missed that one!