Blueforcer/awtrix3 #78: [BUG] AWTRIX Reboot loop when connected to iobroker MQTT server

Beast-Panik, 2023-04-22:
# Bug report

## Describe the bug

When I connect AWTRIX Lite 0.57 to my IOBroker MQTT server, the uhr goes into a bootloop. It appears on the clock briefly "Check" and then "no Up :(" and then comes a restart, until I turn off the MQTT server. I think this is not a wanted behavior?!

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: [ 0.57 ]

## To Reproduce

Steps to reproduce the behavior: connect to the IOBROKER - MQTT server.

Blueforcer, 2023-04-22:
IOBroker usually send its last state on connect, for weird reasons. So also a reboot command.
Turn it off in your broker for awtrix.

Beast-Panik, 2023-04-22:
Unfortunately, it makes no difference what was in the variable to the value reboot. the clock always restarts. 
I have tried the values off,flase,0 and (null) but the clock still always restarts.
[image]

[image]

Blueforcer, 2023-04-22:
With doUpdate and reboot, awtrix doesnt listen to the payload, it starts when this topic was received.
Just turn off "Publish own states on connect" in your broker settings.

closed 2023-04-22 by Blueforcer

Beast-Panik, 2023-04-22:
If I turn it off publishing for stats on connect and subscribe when connecting it works. However, this then applies to all devices that run via mqtt and that makes no sense. For me this is a bug or missing feature. Very sad because the firmware makes a better impression than the PixelIT, but then I will try this one. Thank you.

Blueforcer, 2023-04-22:
But this is not how MQTT or awtrix works, its how IObroker works.
In your case awtrix will then receive every topic wich was subscribed. Even there is no single data.
That make no sense. There are many users using Awtrix with IObroker without problems.

Beast-Panik, 2023-04-22:
This is quite possible that the MQTT server in IOBroker runs differently than mosquito for example. However, I have devices that MQTT should get the current status when connecting, so I do not want to disable the function for AWTRIX. Maybe Mosquito can do this for individual topics but the MQTT server in IOBroker unfortunately cannot. As far as AWTRIX is concerned, it still makes no sense to me if it restarts or checks for an update without me telling it via MQTT. But I'm not really fit as far as mqtt is concerned, if this violates the standards of mqtt it's only a problem of IOBroker, if others can switch off the publishing on connection to make it work that's OK but I'm currently afraid that this could have unwanted effects on other components. Therefore, this is for me currently no solution and I try first again PixelIT even if the everything else is not running smoothly. If that does not work I have to think about something else or just sell the watches again. Nevertheless, thanks for the information, I have at least learned something more ;)

Blueforcer, 2023-04-22:
If you're not satisfied with the awtrix clones you're still welcome to open a feature request.

Beast-Panik, 2023-04-22:
Lol learned something again. Then it makes no sense to install pixelit again. A feature requests only because of my problem I find too much of a good thing. I noticed earlier that the status of the buttons was not updated because the function was disabled. I think that usually does not happen because you would have to press just when the clock goes off but still it irritated me briefly. 
If the others are only forks of AWTRIX I save my time. Eventually I will look at mosquito, maybe I can configure individual topics and disable the publish when connecting only for the awtrix topic.

Blueforcer, 2023-04-22:
They are no forks. They just born from my awtrix2 years ago. But has nothing to do with awtrix light, besides the ideas and some line of codes.
I dont use IObroker so its a bit hard vor me to understand. Maybe you ask the iobroker users in my discord channel how they do it. Otherwise i have no problem with feature requests, if its help. but i need to know how exactly it should work, and no other systems have problems with it.

udo1toni, 2023-04-22:
I'm not familiar with IOBroker, but do you get the topic if connecting with a mqtt sniffer like MQTT Explorer?

There is an option in mqtt, called **RETAIN**. This is part of the protocol and a **retain=true** when sending a payload to a specific topic will result in the behavior described above.  

BUT. 
**Retain=true** should not be used for regular commands - and certainly never for a reboot command, that makes no sense.

I'm pretty sure that the "IOBroker mqtt broker" will accept a delete command from MQTT Explorer to "erase" the retained topic, it's in the specs of mqtt.