Blueforcer/awtrix3 #81: [FEATURE REQUEST] Expose running timer as sensor in Homeassistant

schneekluth, 2023-04-23:
# Feature Request: Expose running timer as sensor in Homeassistant

### Is your feature request related to a problem?
No

### Describe the feature you'd like
Currently when you set a timer on the pixelclock this information is locked and hidden in the clock. It would be nice to have the remaining time exposed as a sensor in Homeassistant.

### Additional context
If the timer information would be exposed I could write an automation to create a custom app that shows the remaining time to show it until timer is finished.

Blueforcer, 2023-04-28:
i have some concerns about sending a message to HA and back to awtrix every second . Wouldn't it be easier if you count the time yourself in homeassistant? you know when you start it.

schneekluth, 2023-04-28:
You're probably right. Way too many messages...

Would it be possible to display remaining time of a timer as custom app? App should only be visible when a timer is running.

Blueforcer, 2023-04-28:
unfortunately the ticker library also doesnt expose the remaining time.
For your purpose i think its better to create a timer custom app from your smarthome. Sending the remaining time each second in only one direction is not a problem.

closed 2023-04-28 by Blueforcer