Blueforcer/awtrix3 #84: [BUG] The lifetime setting sometimes causes the clock to reboot

smarthomejunkie, 2023-04-23:
# Bug report

## Describe the bug

When the lifetime setting is set for a custom app, every now and then the clock reboots. The behavior seems random but happens as soon as you show a Home Assistant sensor that changes its value frequently. Every time when a sensor value updates, a new MQTT.publish command is sent to the clock. I suspect that something is going wrong when the mqtt.publish command is sent for a custom app that has a lifetime set. I've tested this extensively and the issue does not occur as soon as I set the lifetime to 0 for all the custom apps. As soon as I set the lifetime to, let's say, 3 seconds, the device starts to reboot every now and then. The reboot interval is random

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.57

## To Reproduce

Steps to reproduce the behavior:
Set a lifetime of a couple of seconds to one or multiple custom apps and make sure that the custom app gets refreshed as soon as the value of the custom app. Also, switch to that specific app when the value changes, so that it is visible immediately.

So:
1. Publish the app
2. Switch to the just published app
3. Update the published app as soon as the value changes
4. Switch to the just published app

## Expected behavior

A clear and concise description of what you expected to happen.
The clock will not reboot anymore when the lifetime setting is set to a value higher than 0

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

zkrystian, 2023-04-25:
It's going to be more often. 2-3 days ago reboot was 1-2 / day. 1-2 days ago its 2-3 times / day. Today it's for sure 4-5 or more. When this will be fixed?

luchtgitaar, 2023-04-25:
I can't reproduce it all the time but it also happens sometimes when two (or more) mqtt messages are being sent at the same time.

Blueforcer, 2023-04-25:
I'm currently trying to reproduce it. Until there is a fix available, just don't use the lifetime key.

smarthomejunkie, 2023-04-25:
If you are using my Blueprints, make sure you've updated to the latest version. I removed the lifetime option in that version so that the reboot issue won't occur.

commit d25b4327 (2023-04-27): V0.58

- Fixes crashs with lifetime key. closes #84
- Adds dev option to rotate the screen upside down. closes #82
- Allows to play a sound at any time via API
- Adds the possibility to change the textcolor of all native apps individually via settings API. closes #90
- More checks wether to disable automatic transistion. closes #83
closed 2023-04-27 by Blueforcer