Blueforcer/awtrix3 #85: [BUG?] Custom app can't be added again with same name

luchtgitaar, 2023-04-23:
# BUG Report

## Describe the bug
When I remove an app (testapp) with mqtt and try adding the app again with the same name the app does not show up.

## Additional information
I don't have much experience with Awtrix, so maybe I just set up my code wrong, but here it goes. My goal is to show a 'Currently Playing' app, showing the song I play (on Spotify) as an app.  And don't show the app when nothing is playing. 

So my approach is: 

1. Publish app when media.player status changes to playing or changes song
2. Remove app when media.player is paused or idle

I am using:

- Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
- AWTRIX-LIGHT version: v0.57

## Steps to reproduce

1. Publish app via mqtt:
`topic: awtrix/custom/testapp`
` {"text" : "{{ state_attr('media_player.spotify','media_artist') }} -
    {{ state_attr('media_player.spotify, 'media_title') }}", 
     "icon" : 647
     }`
2. Remove app via mqtt
`topic: awtrix/apps`
`   [ {
      "name":"testapp",
      "show":false
   } ]`
3. Publish app again via mqtt:
`topic: awtrix/custom/testapp`
` {"text" : "{{ state_attr('media_player.spotify','media_artist') }} -
    {{ state_attr('media_player.spotify, 'media_title') }}", 
     "icon" : 647
     }`

## Expected behavior
App 'testapp' is published again and shows up back in the app cycle.

## Actual behavior
App does not show up in app cycle. And can't be published again under the same name.

Blueforcer, 2023-04-23:
You set show to false. What is the expected behavior?

I recommend to remove the app by just sending an empty payload to the same topic. It will be completely removed and you can send it at any time again.

luchtgitaar, 2023-04-23:
Thanks. That is a much better approach. And is indeed working. Sorry for fake bug report.

closed 2023-04-24 by Blueforcer