Blueforcer/awtrix3 #87: [BUG] Power API does not wake up display

Adrian-Grimm, 2023-04-25:
# Bug report

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v.0.57

## To Reproduce

Steps to reproduce the behavior:

POST to /api/power with turn off.
POST to /api/power with turn to turn on.

## Expected behavior

It should wake up the display.

## Additional context

It's possible to wake up the display by double click the clock on the button in the middle.

Blueforcer, 2023-04-25:
Please not the Payload/body
https://blueforcer.github.io/awtrix-light/#/api?id=turn-display-on-or-off

Adrian-Grimm, 2023-04-25:
Oh OK, but was the JSON variant not even cleaner?

```
{
   "state": 1
}
```

I just would like to keep it consistent..

Blueforcer, 2023-04-25:
Matter of taste, some users want it this way now 🤣. But I'm still in dev state :)

closed 2023-04-25 by Blueforcer