Blueforcer/awtrix3 #94: [FEATURE REQUEST] Other language support

dalv666, 2023-04-29:
# Feature Request

### Is your feature request related to a problem?

No

### Describe the solution / feature you'd like

I would like to request adding support for Russian in your application. This would require drawing all the symbols for the language and integrating them into the current code. I am willing to assist with the drawing all the symbols. Is it possible to support another language?

### Describe alternatives you've considered

-

### Additional context
-

Blueforcer, 2023-04-29:
Where you need another language?
For the Webinterface?

dalv666, 2023-04-29:
Ability to send characters to display on the watch.

Blueforcer, 2023-04-29:
the awtrix font only supports latin characters. There are no plans to extend this from my side,  because changing fonts is a very huge work.
 If you're able to do it, please feel free to open a PR.

https://github.com/Blueforcer/awtrix-light/blob/main/lib/MatrixUI/Fonts/AwtrixFont.h

closed 2023-04-29 by Blueforcer