Blueforcer/awtrix3 #101: [BUG] Update button in Home Assistant does not work anymore

smarthomejunkie, 2023-04-30:
# Bug report

## Describe the bug

The update button in Home Assistant does not work anymore.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.59

## To Reproduce

Steps to reproduce the behavior:

## Expected behavior

A clear and concise description of what you expected to happen.

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

smarthomejunkie, 2023-04-30:
What I mean is that that button doesn't even work if you use and older firmware and want to update to v1.59. Multiple users mentioned it to me.

closed 2023-04-30 by Blueforcer

luchtgitaar, 2023-05-10:
@smarthomejunkie : Sorry for bumping this already closed thread, but is the update button suppose to update the firmware via OTA? Because that seems not working for me. I have to manual update.

luchtgitaar, 2023-05-12:
> Pressing the 'start update' button in Home Assistant indeed doesn't work anymore. I had to update my device using the physical buttons on the clock.
> 
> [image]

How do you update using the buttons?

smarthomejunkie, 2023-05-12:
I tested this today to update to version 0.62 from 0.61. Updating using the button in HA went real slow and in the end, the device was not updated. Updating using the menu on the clock itself works well.

smarthomejunkie, 2023-05-12:
> > Pressing the 'start update' button in Home Assistant indeed doesn't work anymore. I had to update my device using the physical buttons on the clock.
> > [image]
> 
> How do you update using the buttons?

Long press the middle button to enter the menu.