Blueforcer/awtrix3 #102: [FEATURE REQUEST]

ShayGus, 2023-05-01:
### Describe the solution / feature you'd like

Restore the state of the clock after power cycle. 

When turning on or off apps, this state won't be preserved after reseting the clock. 

For example if I turn off all except for clock,  all apps will be shown again after reset. 

### Describe alternatives you've considered

Manually turning on or off the apps on reset.

Blueforcer, 2023-05-01:
This would cause the ESP to write on the SPIFFS on each app update. Most Clocks will be dead after 2-3 Month. 

But there is absolutely no need to save the apps. Just send your apps as retained, and the mqqt broker will send the last app state on each connect.

Blueforcer, 2023-05-01:
The apps API isn't official available anymore.

Use the onscreen menu to deactive the apps, this will be saved.

ShayGus, 2023-05-01:
I see, thanks.

closed 2023-05-01 by Blueforcer