Blueforcer/awtrix3 #105: [BUG]

glenmich, 2023-05-03:
# Bug report

I have installed the V0.60 software on both of my Ulanzi clocks and since then the display keeps blacking after boot.

## Describe the bug

After installing the V0.60 firmware on a new clock and updating it to an other clock via ota-update, none of them are working anymore. If i reboot them, they'll show the awtrix logo, ip address and the time for a blink of an eye but then the display turns black and doesn't react to anything (mqtt-notify texts, or any button on the clock itself).
However i can still change settings via the browser and the device keeps logged in in my network and home assistant.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: [ eg. v0.60 ]

## To Reproduce

Steps to reproduce the behavior:
Boot the clock

## Expected behavior

A clear and concise description of what you expected to happen.

Cycle trough the menus after booting

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

Blueforcer, 2023-05-03:
Never heard of this problem.
I assume the brightness was set to 0, or simply switched off with an automation.
make sure you down send mqtt messages to brightness or power with retained option

glenmich, 2023-05-03:
[image][image]

I did some restarts and got no pixel to light up at all. Before updating to V0.60 everything worked flawless.

I'll try to reinstall the older firmware now again i guess

Blueforcer, 2023-05-03:
In your screenshot, matrix is switched off.

glenmich, 2023-05-03:
Yes, i had switched it on and of a couple of times. [image]

It seems a bit strange to me that matrix does not turn blue when i toggle the switch. Is this normal?

It's definitely not normal, that the sensors output 0 lux and 0 degree while the box is placed in front of a window in my office

glenmich, 2023-05-03:
[image]

Blueforcer, 2023-05-03:
The bulb is not lit. Click on it and check the brightness.

What firmware have you flashed? Since 0.60 there are two different versions for different hardware

glenmich, 2023-05-03:
[image]
It was turned on at 47% trough the auto brightness but switching to manual and setting 100% white/color did not help either.

That's a nice tip! I have downloaded both and don't know anymore which one i took when updating. Since the interface works perfectly fine i'll try to update the other one :)

Blueforcer, 2023-05-03:
If you're using the ulanzi clock, flash the ulanzi_firmware.bib

glenmich, 2023-05-03:
Well, that was an easy fix.
Now everything works just fine again!

I am sorry to bother you with my stupidity and thank you for the great work you did ! 😊