Blueforcer/awtrix3 #108: [BUG] Incorrect HTTP Response: Header Content-Type is OK

dalv666, 2023-05-06:
# Bug report

## Describe the bug
When call https://github.com/Blueforcer/awtrix-light/blob/main/docs/api.md#colored-indicators endpoint
```
http://[IP]/api/indicator1
```
Server response with Header
```
Content-Type: OK
```

OK is not correct mime type - https://developer.mozilla.org/en-US/docs/Web/HTTP/Basics_of_HTTP/MIME_types

Expected Content-Type
```
Content-Type: text/plain
```

Or better if it response with json body with 
```
Content-Type: application/json
```

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: [ eg. v59 ]

## To Reproduce

Call
```
http://[IP]/api/indicator1
```
Check Header

## Expected behavior

```
Content-Type: text/plain
```

## Screenshots

[image]

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

dalv666, 2023-05-06:
Other endpoints have the same

Blueforcer, 2023-05-09:
https://github.com/Blueforcer/awtrix-light/commit/dc3f124c2810448f0825d3e9002c886f9fc1fb51

closed 2023-05-09 by Blueforcer