Blueforcer/awtrix3 #113: Feature Request - Individual MQTT message for button presses

puca11, 2023-05-10:
**Is your feature request related to a problem?**
Current key presses report 1 <CR> 0 for all keys in MQTT so it's not possible to know which key has been pressed in home automation software, node-red etc..

**Describe the solution / feature you'd like**
Each key press would report individual response so maybe {"button": "left"}, {"button": "middle"}, {"button": "right"}

Blueforcer, 2023-05-10:
The topic tells you wich button was pressed

[image]

puca11, 2023-05-10:
Ahh, ok, thanks.

Blueforcer, 2023-05-10:
glad to help you. Please open another issue if you have further problems.