Blueforcer/awtrix3 #114: [FEATURE REQUEST] Disable transition when one app

dalv666, 2023-05-10:
# Feature Request

### Is your feature request related to a problem?

When one application remains, the transition remains from this application to it.

### Describe the solution / feature you'd like

Check if the application is left alone, do not make the transition.

Blueforcer, 2023-05-11:
Normally awtrx checks and set the transision after each loop change.
Tell me how to reproduce it.

dalv666, 2023-05-11:
`curl --location 'http://192.168.50.215/api/apps' \
--header 'Content-Type: application/json' \
--data '[
    {
        "name": "time",
        "show": true,
        "pos": 0
    },
    {
        "name": "date",
        "show": false,
        "pos": 1
    },
    {
        "name": "temp",
        "show": false
    },
    {
        "name": "hum",
        "show": false
    },
    {
        "name": "bat",
        "show": false
    }
]'`
And then

`curl --location 'http://192.168.50.215/api/apps' \
--header 'Content-Type: application/json' \
--data '[
    {
        "name": "time",
        "show": true,
        "pos": 0
    },
    {
        "name": "date",
        "show": false
    },
    {
        "name": "temp",
        "show": false
    },
    {
        "name": "hum",
        "show": false
    },
    {
        "name": "bat",
        "show": false
    }
]'`

closed 2023-05-11 by Blueforcer

dalv666, 2023-07-05:
I LOVE YOU!

commit b930066d (2023-05-11): V0.62

- Adds AWTRIX Flows link to webinterface
- Inidkator blinking is now set by a given blinking-interval
- Adds LookingEyes app as native app, wich is well known from Awtrix2
- Allows to set a longer Apptime wich is now set as seconds instead of milliseconds. closes #106
- Limits battery percentage. closes #120
- Disables Artnet reading in AP-Mode. closes #116
- Add another transistion check. closes #114
- increase jsonbuffer for drawing commands. closes #112