Blueforcer/awtrix3 #117: OTA upgrade from 0.51 to 0.61 caused display to go black and no response from right button or center button [BUG]

deepcoder, 2023-05-10:
Did OTA firmware upgrade from 0.51 to 0.61 and display went black. Device still working, MQTT setting retained after upgrade and device reported on MQTT. Could not use any API commands to turn on display or change brightness. Also, MQTT topic only reported left button press, center and right buttons did not report in MQTT topic.

Reflashed using Web and USB to 0.61, did device erase during flash. This seems to returned device to showing display.

Blueforcer, 2023-05-11:
You have now skipped 10 versions with many breaking changes during development. That can happen sometimes.

closed 2023-05-11 by Blueforcer

Calimerorulez, 2023-05-11:
I had the same issue going from 0.60 to 0.61. It's unclear to me with file to flash of the two .bin files. I flashed the upgrade.bin and the screen went black. Flashing the ulanzi.bin solved my problem.

Blueforcer, 2023-05-11:
Of course, if you have a ulanzi clock. flash the ulanzi firmware.
If you have an old awtrix 2 hardware use awtrix_upgrade