Blueforcer/awtrix3 #124: [FEATURE REQUEST]

urbatecte, 2023-05-12:
# Feature Request
To configure display speed.

### Is your feature request related to a problem?
Nope !

If so, then add a clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

### Describe the solution / feature you'd like

Add a parameter to configure the spedd of horizontal display. A ratio should be great. 50% for slow reading people like me and 150% for ? Roswell kids :-)

### Describe alternatives you've considered

See you github and @smarthomejunkie one but didn't fing anything.

### Additional context

Add any other context or screenshots about the feature request here.

Blueforcer, 2023-05-12:
It's already possible by changing the FPS

closed 2023-05-12 by Blueforcer

urbatecte, 2023-05-12:
Ok, my bad !
But everything is then slowed down (including the eyes :-()
Therfotre, should it be possible to configure a 'FPS' for dedicated message (eg: long one)
Thanks, and thanks for your amazing firmware !