Blueforcer/awtrix3 #125: Set scroll speed for oversized text including start scroll delay for custom apps

phoenixswiss, 2023-05-13:
# Question

- Is there a way to set the scroll speed of oversized custom app text? 
- Could we even set a start delay for oversized custom text?

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.63

## Additional context

The text scrolls too fast and it is sometimes difficult to read the digits of a sensor value.

Blueforcer, 2023-05-13:
Currently the scroll speed is handled by the given FPS.

You cannot set a delay but you can set a start offset so the text starts scrolling at position 32 or how you like it

phoenixswiss, 2023-05-13:
Perfect, thank you for the fast reply.

It works as expected.

closed 2023-05-13 by Blueforcer