Blueforcer/awtrix3 #128: [BUG] Issue Downloading Icons

blindguynar, 2023-05-16:
When Running Awtrix-Light on Ulanzi Clock i'm unable to download icons.  The Icons preview fine, but when clicking download i always receive an error no matter the icon choice.  Awtrix Icon's have exact same behavior.  

i can manually upload an icon in the files area under ICONS and is fine.  This icons download just isn't working.

I do not see where logs are stored, but if directed i will provide them.

This is the preview
[image]

This is the error
[image]

Blueforcer, 2023-05-16:
Logs can be found in the browsers console

blindguynar, 2023-05-16:
Here is the error in the chrome Console log.  I'm on a macbook pro in case that matters.

[image]

blindguynar, 2023-05-16:
Ok i need to close this issue.  This is something specific to my machine.  I tried it on my other macbook pro and it's fine.  I've got some weird website filtering going on or something with this machine.  Sorry this is NOT a bug with awtrix-light.

closed 2023-05-16 by blindguynar