Blueforcer/awtrix3 #129: [FEATURE REQUEST] Overwrite notifcations

berrywhite96, 2023-05-16:
# Feature Request

### Is your feature request related to a problem?

Updated from 0.6 to 0.6.5 and now my notifications are stacked and shown one after the other. In 0.6 the notifications were overwritten.

### Describe the solution / feature you'd like

A new attribute for the notification, something like `overwritable` which indicates if the notification can be overwritten by a newer one.

blindguynar, 2023-05-17:
this can be done within Home Assistant, Node-Red, IFTTT, or any other host of flow based Automation platforms.  

I have it behaving as you suggest in HA by way of creating an automation that is basically running as a state machine for notifications.  All clock notifications trigger that state machine automation and if a certain notification is there no others will display, but other types can overwrite each other.

it might be the largest automation i've ever written.  i need to convert it to a script honestly.  it just kept GROWING!

luchtgitaar, 2023-05-17:
Can't you just create a "dismiss notification" action in every notification automation as first step?

berrywhite96, 2023-05-17:
> Can't you just create a "dismiss notification" action in every notification automation as first step?

That would be possible, but sometimes it will send about 4-5 notifications in a second. So I want to minimize the amount of request sent to the device.

berrywhite96, 2023-05-17:
> So i will add a new key "stack" wich is true in standard. if you set this to false, the current one will be replaced immediately by the new one? Is that the way to go?

Yes, this would be perfect.

commit d75fde41 (2023-05-17): V0.66

- Allows to decide if a notification is stacked. closes #129
- Add indicator3 to http API. closes #130
- Adds some meaningful responses to http api. closes #131
- Fixes a bug where customapp crashes without an icon. closes #132