Blueforcer/awtrix3 #137: [FEATURE REQUEST] API for setting an array of apps that shall be displayed

i-C-o-d-e-r, 2023-05-19:
# Feature Request

### Is your feature request related to a problem?

I don't want to display some built-in apps. For example I don't want to display the eyes app or the battery app. However I want to keep the MQTT data for the battery for my home assistant. So my feature request is just about skipping/hiding some of the built-in apps, not switching them off completely (at least for the battery).

I think this can be useful for custom apps as well, when you temporarily want to hide/skip custom apps.

### Describe the solution / feature you'd like

Provide an API to set an array of apps (built-in and maybe even custom) that shall be displayed. The index of the app in the array determines the display order when cycling through the apps.

For example

POST http://[IP]/api/apps with JSON
```
{
   "apps": [
    "time",
    "mycustom1",
    "date",
    "mycustom2"
   ]
}
```

would display "time", "mycustom1", "date" and "mycustom2" only and exactly in this order.

### Describe alternatives you've considered

I read the documentation and found out that as a "Hidden feature" it is possible to disable the reading of humidity and temperature. But this is not what I want. I want humidity and temperature to be still read (for MQTT, so it can be evaluated by my home assistant instance), but not displayed on the device.

And it is limited to those two "apps" only...

### Additional context

None

Blueforcer, 2023-05-19:
Something similar exists some versions ago, but causes more questions as it helps.
Its not documented but you can still use it
https://github.com/Blueforcer/awtrix-light/blob/e77e875bad1ef7adae63952354f39c1829cc801f/docs/api.md#addremove-and-rearange-apps

i-C-o-d-e-r, 2023-05-20:
Thanks for that hint. I looked at the code and I understand that this may cause questions. I will give it a try and see what I can do.

If I find a good solution, would you accept a PR for this?

Blueforcer, 2023-05-20:
Solution?
this API endpoint (**/apps**) is still active

i-C-o-d-e-r, 2023-05-20:
I understood that the endpoint is still active.

> but causes more questions then it helps.

I was referring to your statement that it is undocumented, because it causes "questions". Maybe I can find a silightly different approach that causes no "questions/issues" (and maybe I can't). But I don't want to waste time, if you would not accept a PR for it.

Blueforcer, 2023-05-20:
Sure, I'm happy to get a PR

closed 2023-06-01 by Blueforcer