Blueforcer/awtrix3 #143: [BUG]

RDDJ, 2023-05-29:
# Bug report
New TC001 functioning slowly after Awtreic flashing

## Describe the bug
After flashing my second, brand new TC001 with Awtric Light, it is functioning slowly and not connecting to home assistant. The buttons don't respond immediately when pushed and the flashing colon on the clock is also delayed on seemingly every-other flash.

## Additional information
- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.66

## To Reproduce
Reflashed multiple times and getting the same issue every time

## Logs
```text
E (521) esp_core_dump_flash: No core du� �partition found!
E (521) esp_core_dump_flash: No core dump partition found!
[  2078][E][vfs_[2080] [setup]: Setup periphery
[2240] [loadSettings]: Webserver configuration loaded
[2241] [setMatrixLayout]: Set matrix layout to 0
Framebuffer_GFX::begin Width: 32 Height: 8 Num Pixels: 256
Connecting to T-800
.................[8000] [setup]: My IP: 192.168.101.41
[11345] [setup]: Webserver loaded
[15605] [setup]: Starting Homeassistant discovery
[15608] [connect]: Connecting to MQTT with login
[ 43024][E][WebServer.cpp:649] _handleRequest(): request handler not found
[ 43028][E][vfs_api.cpp:105] open(): /littlefs/api/VvLNyvlvLWJc0pOr-erql9VpyO2wqbG1ZAblTfv8/lights/11.gz does not exist, no permits for creation
[ 43035][E][vfs_api.cpp:105] open(): /littlefs/api/VvLNyvlvLWJc0pOr-erql9VpyO2wqbG1ZAblTfv8/lights/11 does not exist, no permits for creation
E (521) esp_core_dump_flash: No core du� �partition found!
E (521) esp_core_dump_flash: No core dump partition found!
[  2078][E][vfs_[2080] [setup]: Setup periphery
[2240] [loadSettings]: Webserver configuration loaded
[2241] [setMatrixLayout]: Set matrix layout to 0
Framebuffer_GFX::begin Width: 32 Height: 8 Num Pixels: 256
Connecting to T-800
................[7698] [setup]: My IP: 192.168.101.41
[11043] [setup]: Webserver loaded
[12652] [setup]: Starting Homeassistant discovery
[12655] [connect]: Connecting to MQTT with login
[ 20080][E][WebServer.cpp:649] _handleRequest(): request handler not found
[ 20084][E][vfs_api.cpp:105] open(): /littlefs/api/VvLNyvlvLWJc0pOr-erql9VpyO2wqbG1ZAblTfv8/lights/39.gz does not exist, no permits for creation
[ 20091][E][vfs_api.cpp:105] open(): /littlefs/api/VvLNyvlvLWJc0pOr-erql9VpyO2wqbG1ZAblTfv8/lights/39 does not exist, no permits for creation
[ 31168][E][WebServer.cpp:649] _handleRequest(): request handler not found
[ 31171][E][vfs_api.cpp:105] open(): /littlefs/api/VvLNyvlvLWJc0pOr-erql9VpyO2wqbG1ZAblTfv8/lights/18.gz does not exist, no permits for creation
[ 31178][E][vfs_api.cpp:105] open(): /littlefs/api/VvLNyvlvLWJc0pOr-erql9VpyO2wqbG1ZAblTfv8/lights/18 does not exist, no permits for creation
[ 50102][E][WebServer.cpp:649] _handleRequest(): request handler not found
[ 50106][E][vfs_api.cpp:105] open(): /littlefs/api/VvLNyvlvLWJc0pOr-erql9VpyO2wqbG1ZAblTfv8/lights/48.gz does not exist, no permits for creation
[ 50113][E][vfs_api.cpp:105] open(): /littlefs/api/VvLNyvlvLWJc0pOr-erql9VpyO2wqbG1ZAblTfv8/lights/48 does not exist, no permits for creation
```

## Additional context
Updated my other TC001 to 0.66 without issue. Could potentially be hardware, but need to eliminate software as well.

Blueforcer, 2023-05-29:
Sounds like awtrix simply cannot connect to the mqtt Broker, and always tries it again. Since the connection is not asynchronous the software stops for this time.

Check your mqtt settings.

RDDJ, 2023-05-29:
Thanks for the pointer; If I remove the mqtt IP then the buttons respond and the clock colon flashes as expected, so looks like you're correct. Appreciate the help.

closed 2023-05-30 by Blueforcer