Blueforcer/awtrix3 #145: [BUG] Matrix does not power on for MQTT Notification

jeremywillans, 2023-05-31:
# Bug report

## Describe the bug

Matrix does not power on when receiving a notification via MQTT.

However, if sound is included in payload, this is played (with matrix off)

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.66

## To Reproduce

1. Using MQTT, publish power off command
2. Using MQTT, publish notify command
3. Sound (if included) will play.

## Expected behavior

Matrix should turn on and display notification

Blueforcer, 2023-05-31:
Isn't it a feature request?

jeremywillans, 2023-05-31:
Well, the sound plays, but the screen does not turn on.

Blueforcer, 2023-05-31:
Because you turned the matrix off, not the sound

Blueforcer, 2023-05-31:
Because you turned the display off, not the sound 😅

jeremywillans, 2023-05-31:
Fair enough! I have updated this to reflect a feature request - be good to see it wake when receiving a notification. 👍

Blueforcer, 2023-05-31:
I think it would be better to deactive the sound too. And make the wakeup optional via jsonkey.

commit 2af991eb (2023-06-01): 0.67

- Adds wakeup command to the notification API. closes #145
- Increase dev json buffer. closes #147
- fixes a bug with mulitple pages command. closes #146
- clear json buffer after modifing the apps vector.
- adds hostname
- adds new HA sensor to get the mqtt prefix
- fixes a bug where alarms doesnt work
- Switch http api returns an error, if the app doesnt exist. closes #140
closed 2023-06-01 by Blueforcer