Blueforcer/awtrix3 #148: [BUG] Matrix not lighting up after update to 0.66

cebrac, 2023-06-01:
I upgraded from 0.62 to 0.66 firmware via web interface upload button.
Matrix was showing time at that point. Web interface said device rebooting but matrix did not show the usual indications of a reboot. Time display remained. Didn't check whether time was correct, did not pay attention to it.

After reboot, device became accessible through MQTT and web interface but matrix remained displaying "a" time. Tried two button reboot, now the matrix stays dark.

Device boots, beep can be heard. Seems like just the matrix isn't initialized properly. Will let battery run dry so there is a proper "cold" restart and then report further.

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT 0.66

## To Reproduce

Unclear whether this can be forced. If this is likely to be a hardware issue, this ticket can be closed.

## Expected behavior

Matrix lighting up and displaying version number, as well as other contents upon reboot.

Blueforcer, 2023-06-01:
Do you use IObroker?

cebrac, 2023-06-01:
No I drive the clock via MQTT from a bunch of scripts on a different software platform

closed 2023-06-01 by Blueforcer