Blueforcer/awtrix3 #149: [BUG] Battery data missing from stats since 0.66

cebrac, 2023-06-01:
Since updating my clock to 0.66, the battery percentage is missing from the "stats" json string, via MQTT as well as via the HTTP api.

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version 0.66

## To Reproduce

Update to 0.66 I suppose.

## Expected behavior

Fields "bat" and "bat_raw" should exist in the json object returned by the clock as part of the stats package.

Blueforcer, 2023-06-01:
Seems like you flashed the wrong firmware.
If you have the Ulanzi clock, you need to flash the Ulanzi_firmware.. Not awtrix2_upgrade

Blueforcer, 2023-06-01:
This would also cause that the matrix doesnt work( like your other issue) because each firmware uses different pinouts

cebrac, 2023-06-01:
Can confirm. I'm a moron and this ticket can be closed. Sorry for the trouble and thanks for the help.

closed 2023-06-01 by Blueforcer