Blueforcer/awtrix3 #151: [BUG] Wakeup does not update MQTT/HA

jeremywillans, 2023-06-02:
# Bug report

## Describe the bug

Firstly thanks for the fast turnaround on adding the Wakeup capability!

However MQTT/HA is not updated to reflect this change.

- currentApp topic remains the current App (not Notification)
- Matrix switch in HA does not updated

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.67

## To Reproduce

Steps to reproduce the behavior:
1. Power off Matrix
2. Send Notifiction to device

## Expected behavior

A clear and concise description of what you expected to happen.
1. currentApp topic should update to "Notification"
2. Matrix should switch on

Blueforcer, 2023-06-05:
Updateing Matrix switch is not possible. because its still off internally

closed 2023-06-05 by Blueforcer

jeremywillans, 2023-06-06:
Thanks for resolving - now that it powers back off after the notification is dismissed, I agree this isnt relevant 👍

jeremywillans, 2023-06-06:
@Blueforcer, it looks like the currentApp topic is not restored after the notification is dismissed 😞

Blueforcer, 2023-06-06:
The App name is only updated when the app is switching.

jeremywillans, 2023-06-13:
@Blueforcer, thanks for the response, to better clarify, my use case here is being able to understand the current state of the device, especially when the notification has been dismissed.

In a state of matrix off, the currentApp correctly reflects the shown app (prior to turn off) and then when a notification is issued, its updated to Notification (and matrix state is still off as we discussed earlier). However once the notification is dismissed, IMO currentApp should be set back to its previous state, which can also be achieved when either a) turning on the matrix or b) pressing either of the button, even when the matrix is off.

Essentially without this I have no way of knowing if/when the notification has been acknowledged. Happy for other suggestions?

In other news, my Clock has fallen victim to the clock death plague so i am waiting on further correspondence from Ulanzi support.

Blueforcer, 2023-06-13:
Matrix Off, doesnt mean Awtrix Off.
It just turn the brightness down to zero but awtrix still runs normaly.
You have two ways to to get the acknowledge:
1. Just trigger if current app switches from notification to something else.
2. If current app = notification AND middle button is pressed

commit cc749a48 (2023-06-05): v0.68

- You can set hum and temp offset via dev.json. closes #154
- Change awtrix font to use less pixel for degree symbol.
- Adds UTF8 encoding to text fragments and text drawing
- Restrore matrix state after notification dismiss when using wakeup. closes #152
- Wakeup updates current appname to Notification. Closes #151
- Adds more keys to settings API (read) . closes #160
- Sets DefaultDataPrefix for HA discrovery ti the MQTT topic. closes #161
- Adds second dot to DD.MM. date format. closes #164