Blueforcer/awtrix3 #154: [BUG] temperature not correkt

BenAhrdt, 2023-06-02:
The real temperature is 23,5°C and the TC001 displays 32°C
Maybe there can be an Offset?

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: [ eg. v0.66 ]

Blueforcer, 2023-06-02:
Since the sensor measures the inside of the housing, there are many factors that can lead to deviation. The brightness of the matrix, how many pixels light up in which colour and also the outside temperature.
Unfortunately, you will not be able to set the exact temperature unless you change it almost every hour.
I recommend deactivating the native app and using your own custom app with an external sensor.

cebrac, 2023-06-03:
Agreed. The sensor is helpful to early detect a faulty charger/battery behaviour though!

closed 2023-06-05 by Blueforcer

shyney7, 2023-10-09:
> Agreed. The sensor is helpful to early detect a faulty charger/battery behaviour though!

How does the case temperature contribute to detecting a faulty charger?

cebrac, 2023-10-09:
> How does the case temperature contribute to detecting a faulty charger?

Since the actual **charging regulator** resides within the device itself, an increase in temperature might indicate a fault with it, among other possibile problems.

The sensor can't help detect a faulty **psu**, obviously.

commit cc749a48 (2023-06-05): v0.68

- You can set hum and temp offset via dev.json. closes #154
- Change awtrix font to use less pixel for degree symbol.
- Adds UTF8 encoding to text fragments and text drawing
- Restrore matrix state after notification dismiss when using wakeup. closes #152
- Wakeup updates current appname to Notification. Closes #151
- Adds more keys to settings API (read) . closes #160
- Sets DefaultDataPrefix for HA discrovery ti the MQTT topic. closes #161
- Adds second dot to DD.MM. date format. closes #164