Blueforcer/awtrix3 #156: [QUESTION] Recovery and diagnosis options for "half dead" clock?

hagenbuch, 2023-06-02:
I've read the "Clock dead" thread here but my case is different: I ordered two devices and got one flashed to Awtrix using the old softwares "upload". It works fine.

Before I did that, I tried to flash my other device via the online flasher. It seemed to work as while updating, I've read the red "UPDATE" banner and the beeper had been on. The flashing process apparently got over the "erasing" stage and quit with something like "timeout" in a relatively early stage of the flashing process.

I am still able to turn the device off by a long press on "<" and ">" and I can turn it on the same way. Turning on means the screen remains black but I hear a constant beep sound, only with added low volume "ticks" every 0.5 seconds.

The charging lamp is red at the beginning but turns green later.

The web installer still "sees" a serial interface so it can't be completely dead. The installer stops "Failed to initialize. Try resetting your device or holding the BOOT button while clicking INSTALL".

What is the proper reset procedure? 

Can I somehow recover the device with esptool.py or something?

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0,67 (the one that wors)

## Logs

No log available

## Additional context

None

RedFuseCap, 2023-06-03:
I ran into the same problem today. First I had the clock connected to a USB3.0 port on the front of the PC. This gave me the "Failed to initialize" error. Then I connected the clock to a USB port on the back of the PC, same error. I then noticed it was a USB2.0 port. Finally I connected the clock to **a USB3.0 port on the back of the PC**, and then it worked!

Blueforcer, 2023-06-03:
First, dont use the original USB cable.
You can also try another USB port

Blueforcer, 2023-06-05:
@hagenbuch  Is this issue solved?

hagenbuch, 2023-06-06:
Thanks for asking: No, Ulanzi will send me a new device. 

When I tried to only connect it to Linux, these messages showed up:

    [135490.495562] usb 1-1: new full-speed USB device number 14 using xhci_hcd
    [135490.644627] usb 1-1: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice=81.34 
    [135490.644639] usb 1-1: New USB device strings: Mfr=0, Product=2, SerialNumber=0   
    [135490.644645] usb 1-1: Product: USB Serial
    [135490.646833] ch341 1-1:1.0: ch341-uart converter detected
    [135490.647634] usb 1-1: ch341-uart converter now attached to ttyUSB0
    [135493.529746] usb 1-1: usbfs: interface 0 claimed by ch341 while 'brltty' sets config #1
    [135493.530690] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
    [135493.530738] ch341 1-1:1.0: device disconnected 

.. so I could not even try to read the ESP32 chip info..

I am slightly familiar with esptool and have version 3.1 installed here..

DcFamas, 2023-06-06:
@Blueforcer had the same problem. For me it fixed another USB cable and another USB port. Funny that this device is a little bit bitchy, I flashed about 50 ESP32 on this exact USB port on my computer. It would be great to add this tipps and tricks to the Online flasher page for those who are struggling like me. 

Thanks and best regards! :)

Blueforcer, 2023-06-06:
Ok, I now understand that it has nothing to do with the firmware. I am therefore closing this issue.

closed 2023-06-06 by Blueforcer

Blueforcer, 2023-06-06:
Not really, besides using other cables / USB Ports or maybe other Computer there is nothing you can do I think

TheWalkingDadNL, 2023-09-05:
Found this page when searching on google why i can't flash the Ulanzi. Solution for me was not to use different USB port. The solution was to unplug all USB-devics exept my mouse. This way i could click on the button and after that it instantly connected.

SebastianHanz, 2024-07-27:
> I ran into the same problem today. First I had the clock connected to a USB3.0 port on the front of the PC. This gave me the "Failed to initialize" error. Then I connected the clock to a USB port on [...]

Switching from frontpanel USB to backpanel USB 3.x (blue one, orange didn't make it) was also the solution in my case. 
Thank you very much!