Blueforcer/awtrix3 #158: [BUG] Degree symbol not working in 0.67

cebrac, 2023-06-03:
I used to send custom app text as segmented text via MQTT to display the current outside temperature and it would disply the ° symbol as a small square of 4 dots. Since updating two of my clocks to 0.67 the symbol is not displayed any more. Not sure whether this stems from how php seems to be encoding the symbol:

`{"text":[{"t":"14\u00b0","c":"FFFFFF"},{"t":"19\u00b0","c":"884433"}],"color":"FFFFFF","icon":50,"repeat":2}`

This is the MQTT payload. Displays properly on 0.66 but not on 0.67

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version 0.67

## To Reproduce

Publish above payload as a custom app via MQTT.

## Expected behavior

The message being displayed with the ° symbol intact.

SenH, 2023-06-03:
Is it really necessary to encode the symbol, it works just fine without (also on 0.67)
`{"text": "18°C"}`

[image]
"}

cebrac, 2023-06-03:
It's what PHP json_encode() function outputs on a windows machine when I use the ° symbol in a string field.

SenH, 2023-06-03:
You can solve it with the flag `JSON_UNESCAPED_UNICODE`:
```
php > echo json_encode('18°C');
"18\u00b0C"

php > echo json_encode('18°C', JSON_UNESCAPED_UNICODE);
"18°C"
```

cebrac, 2023-06-04:
True. Just leaves me wondering why the behavior has changed from 0.66

Blueforcer, 2023-06-05:
So this issue can be closed?

cebrac, 2023-06-06:
If you don't consider it a problem, sure. I thought that perhaps along with the changed behavior something else might also have been broken. But it can certainly be worked around.

closed 2023-06-12 by Blueforcer