Blueforcer/awtrix3 #161: [FEATURE REQUEST] Change  DefaultDataPrefix="SHA"  to  DefaultDataPrefix=MQTT_PREFIX

robercy, 2023-06-03:
Publish the system data using the MQTT_PREFIX defined on the configuration page, rather than from *DefaultDataPrefix= "SHA". To keep all data together on the same main "Prefix" Topic.

For example, the topics would be:

Prefix/app 
Prefix/bri 
Prefix/btnL
Prefix/temp
....

[image]

closed 2023-06-05 by Blueforcer

robercy, 2023-06-05:
Guys, You Rocks !!!
Keep the great job.

Blueforcer, 2023-06-05:
Still a one-man show, but thanks 😁

robercy, 2023-06-06:
One-man show as good as a Multi-man show.
My reverences to your quality work and dedication.

robercy, 2023-07-25:
Hi, one problem killed, another born.

The awtrix still publish the status to SHA during boot. 
Maybe the software timing is not correct, like starting MQTT connection before load "Prefix" configuration from config.json file.

Thanks, again.

Blueforcer, 2023-07-25:
This should be only once during boot, but this is a problem of the library itself it send something during initializing before I can set the prefix.

After that it's sends to the correct prefix

robercy, 2023-07-25:
It's Ok! 
I allways compile your source code, then I solved this problem, changing "static const char *DefaultDataPrefix" direct on HAMqtt.cpp file.

commit cc749a48 (2023-06-05): v0.68

- You can set hum and temp offset via dev.json. closes #154
- Change awtrix font to use less pixel for degree symbol.
- Adds UTF8 encoding to text fragments and text drawing
- Restrore matrix state after notification dismiss when using wakeup. closes #152
- Wakeup updates current appname to Notification. Closes #151
- Adds more keys to settings API (read) . closes #160
- Sets DefaultDataPrefix for HA discrovery ti the MQTT topic. closes #161
- Adds second dot to DD.MM. date format. closes #164