Blueforcer/awtrix3 #162: [QUESTION] settings TCOL - hex to int?

klein0r, 2023-06-04:
# Question

How does the conversion from hex to int work?

When setting `TCOL` to `#0000FF`, the settings endpoint return `31` for that value (instead of `255`). How does the conversion works?

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.67

## Logs

```text
2023-06-04 11:03:28.409  - debug: awtrix-light.0 (112550) sending "POST" request to "/api/settings" with data: {"TCOL":"#0000FF"}
2023-06-04 11:03:28.458  - debug: awtrix-light.0 (112550) received 200 response from "/api/settings" with content: "OK"
2023-06-04 11:03:28.459  - debug: awtrix-light.0 (112550) sending "GET" request to "/api/settings" without data
2023-06-04 11:03:28.502  - debug: awtrix-light.0 (112550) received 200 response from "/api/settings" with content: {"FPS":23,"ABRI":true,"BRI":255,"ATRANS":true,"TCOL":31,"TSPEED":400,"ATIME":7000,"TFORMAT":"%H:%M:%S","DFORMAT":"%d.%m.%y","SOM":true,"CEL":true,"MAT":0,"SOUND":true,"GAMMA":0}
```

Blueforcer, 2023-06-04:
This is the raw RGB565 value (16-bit color) wich is the converted RGB888 (24-bit color) youve send.
The RGB565 color bitmask is rrrrrggg:gggbbbbb: red - 5 bit (0-31) in MSB, green - 6 bit (0-63), blue - 5 bit (0-31) in LSB.
 
 RGB888 (0000FF) -> RGB565 (001F) = 31 decimal

Blueforcer, 2023-06-04:
Du kannst auch gerne mal den discord server joinen, da können wir für deine Integration in einem eigenen thread evtl enger zusammenarbeiten

closed 2023-06-04 by klein0r