Blueforcer/awtrix3 #163: Bootloop nach eingabe der MQTT Daten

haahne, 2023-06-04:
Guten Tag,
habe die Uhr erfolgreich geflasht. 

Aber immer wenn ich die daten vom MQTT Broker eingebe und anschließend reboote . Steht auf der Uhr nach der Bootanimation : " CHECK NO UP :( "
Danach erfolgt ein Reboot dies hört erst auf wenn ich die IP Vom Broker lösche.  Broker ist der vom ioBroker. 
Habe auch eine 2. Instanz vom MQTT Adapter installiert und einen anderen Port genommen. Bringt leider auch nichts. 

[logfile.txt](https://github.com/Blueforcer/awtrix-light/files/11645485/logfile.txt)

Blueforcer, 2023-06-04:
Du musst "eigene states beim verbinden publizieren" im Broker deaktivieren.
Ansonsten sendet iobroker immer an reboot sobald sich awtrix verbindet

haahne, 2023-06-04:
Habe ich rausgenommen. 
Keine Besserung: 
[image]

MFV, 2023-06-04:
"States beim Subscribe publizieren" muss ebenfalls deaktiviert werden.

haahne, 2023-06-04:
Ja das habe ich eben dann noch versucht. :D 
Vielen Dank

closed 2023-06-04 by haahne