Blueforcer/awtrix3 #164: [BUG] Month missing following dot

b-reich, 2023-06-04:
# Bug report

## Describe the bug

Hi just a minor bug.
changing the date to `DD.MM`. It should be have a following dot there.
`04.06` is not a correct date. It should be `04.06.`

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.67

## To Reproduce

Change Date to `DD.MM`

## Expected behavior

It should be `DD.MM.`

## Screenshots

[image]

## Additional context

Add any other context about the problem here.

closed 2023-06-05 by Blueforcer

Blueforcer, 2023-06-05:
damn youre right. i made a mistake on this commit.
Ive updated the 0.68. please update it manually again.

Blueforcer, 2023-06-05:
Yes. i didnt updated the version name.

Blueforcer, 2023-06-05:
Check again in few minutes. Github need some time for publishing

b-reich, 2023-06-05:
Flashed it over the web interface now. Seems to work :)
Thx

commit cc749a48 (2023-06-05): v0.68

- You can set hum and temp offset via dev.json. closes #154
- Change awtrix font to use less pixel for degree symbol.
- Adds UTF8 encoding to text fragments and text drawing
- Restrore matrix state after notification dismiss when using wakeup. closes #152
- Wakeup updates current appname to Notification. Closes #151
- Adds more keys to settings API (read) . closes #160
- Sets DefaultDataPrefix for HA discrovery ti the MQTT topic. closes #161
- Adds second dot to DD.MM. date format. closes #164